xoseperez/espurna #733: Buttons not reporting on MQTT

joearkay, 2018-03-27:
Hi all - firstly I'd like to say I love the codebase - it's proving extremely useful so far! 

Quick question with the buttons - I'm not having much luck getting their states reported over MQTT. I have two relays configured which report great over MQTT, likewise with the DHT22 I have setup on pin 14. I have configured 3 buttons as follows:
```
      #define BUTTON1_PIN		  16 //D0 - Button1
      #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
      #define BUTTON1_PRESS           BUTTON_MODE_TOGGLE
      #define BUTTON1_CLICK           BUTTON_MODE_TOGGLE
      #define BUTTON1_DBLCLICK        BUTTON_MODE_TOGGLE
      #define BUTTON1_LNGCLICK        BUTTON_MODE_TOGGLE
      #define BUTTON1_LNGLNGCLICK     BUTTON_MODE_TOGGLE
	  
      #define BUTTON2_PIN         12 //D6 - Button2
      #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
      #define BUTTON2_PRESS       BUTTON_MODE_TOGGLE
      #define BUTTON2_PRESS           BUTTON_MODE_TOGGLE
      #define BUTTON2_CLICK           BUTTON_MODE_TOGGLE
      #define BUTTON2_DBLCLICK        BUTTON_MODE_TOGGLE
      #define BUTTON2_LNGCLICK        BUTTON_MODE_TOGGLE
      #define BUTTON2_LNGLNGCLICK     BUTTON_MODE_TOGGLE
	  
	  
      #define BUTTON3_PIN         13 //D7 - PIR INPUT
      #define BUTTON3_MODE        BUTTON_PUSHBUTTON
      #define BUTTON3_PRESS       BUTTON_MODE_TOGGLE
      #define BUTTON3_PRESS           BUTTON_MODE_TOGGLE
      #define BUTTON3_CLICK           BUTTON_MODE_TOGGLE
      #define BUTTON3_DBLCLICK        BUTTON_MODE_TOGGLE
      #define BUTTON3_LNGCLICK        BUTTON_MODE_TOGGLE
      #define BUTTON3_LNGLNGCLICK     BUTTON_MODE_TOGGLE
```

This is all configured within `hardware.h`. I have my relays and DHT configured within the same section, so I'm happy that this config is getting picked up. 

I've also tested the GPIO pins of each 'button' I am using, making sure they go from HIGH to LOW when I click the switch. 

Button 1 (D0) uses a Pullup resistor, so is normally HIGH. Button 2 (D6) is the same. Button 3 (D7) uses a pulldown, so is usually LOW and is connected to the output pin of a 3.3V logic PIR sensor. 

None of the above pushes through via MQTT. Again, I'm happy that my MQTT is configured at both ends, as I can read temp/humidity/relay info. Any suggestions? 

NB - Sorry for using GitHub issues as a place for help - I just couldn't find a forum!

Thanks all.

lobradov, 2018-03-28:
Can you try with `BUTTON{x}_[click/dblclick/lngclick/lnglngclick]` = `BUTTON_MODE_NONE`?

That works for me.

joearkay, 2018-03-28:
Hi @lobradov - I've tried this, with no luck still unfortunately:

```

	  #define BUTTON1_PIN		  16 //D0 - Button1
	  #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
	  #define BUTTON1_PRESS           BUTTON_MODE_TOGGLE
          #define BUTTON1_CLICK           BUTTON_MODE_NONE
          #define BUTTON1_DBLCLICK        BUTTON_MODE_NONE
          #define BUTTON1_LNGCLICK        BUTTON_MODE_NONE
          #define BUTTON1_LNGLNGCLICK     BUTTON_MODE_NONE
	  
	  #define BUTTON2_PIN         12 //D6 - Button2
	  #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
	  #define BUTTON2_PRESS       BUTTON_MODE_TOGGLE
	  #define BUTTON2_PRESS           BUTTON_MODE_NONE
          #define BUTTON2_CLICK           BUTTON_MODE_NONE
          #define BUTTON2_DBLCLICK        BUTTON_MODE_NONE
          #define BUTTON2_LNGCLICK        BUTTON_MODE_NONE
          #define BUTTON2_LNGLNGCLICK     BUTTON_MODE_NONE
	  
	  
	  #define BUTTON3_PIN         13 //D7 - PIR INPUT
	  #define BUTTON3_MODE        BUTTON_PUSHBUTTON
	  #define BUTTON3_PRESS       BUTTON_MODE_TOGGLE
	  #define BUTTON3_PRESS           BUTTON_MODE_NONE
          #define BUTTON3_CLICK           BUTTON_MODE_NONE
          #define BUTTON3_DBLCLICK        BUTTON_MODE_NONE
          #define BUTTON3_LNGCLICK        BUTTON_MODE_NONE
          #define BUTTON3_LNGLNGCLICK     BUTTON_MODE_NONE
```

I know that I'm getting MQTT messages through, as I can see data for the topics `rssi`, `uptime`, `vcc` and `status` etc. I'm using the `+` wildcard so I can subscribe to every topic from this board - however still no data when I try the buttons. I've hooked my multimeter up to the pins to ensure they go high/low when I press the button, but still get nothing. 

Thanks, 
Joe

joearkay, 2018-03-28:
This should also explain most of the settings that have been configured via the UI. Nothing button-specific has been setup there (no option to, is there?):

```
{
  "app": "ESPURNA",
  "version": "1.12.4",
  "adminPass": "----------",
  "alexaEnabled": "1",
  "apiEnabled": "0",
  "apiKey": "DD489703881472B3",
  "apiRealTime": "0",
  "board": "2",
  "boardName": "NODEMCU_LOLIN",
  "btnDelay": "500",
  "btnGPIO0": "0",
  "btnRelay0": "0",
  "cfg": "3",
  "dczEnabled": "0",
  "dczMagnitude0": "0",
  "dczMagnitude1": "0",
  "dczRelayIdx0": "0",
  "dczRelayIdx1": "0",
  "dczTopicIn": "domoticz/in",
  "dczTopicOut": "domoticz/out",
  "dns0": "8.8.8.8",
  "energyUnits": "0",
  "haEnabled": "0",
  "haPrefix": "homeassistant",
  "hostname": "ESPURNA_0FADC5",
  "idbEnabled": "0",
  "ledGPIO0": "2",
  "ledLogic0": "1",
  "ledMode0": "1",
  "mask0": "255.255.255.0",
  "mqttClientID": "wifihomenode0",
  "mqttEnabled": "1",
  "mqttGroup0": "wifihome/wifihomenode0/relay0",
  "mqttGroup1": "wifihome/node0relay1",
  "mqttGroupInv0": "0",
  "mqttGroupInv1": "0",
  "mqttKeep": "30",
  "mqttPassword": "",
  "mqttPort": "1883",
  "mqttQoS": "2",
  "mqttRetain": "1",
  "mqttServer": "192.168.1.189",
  "mqttTopic": "wifihome/wifihomenode0",
  "mqttUseJson": "0",
  "mqttUseSSL": "0",
  "mqttUser": "wifihome",
  "nofussEnabled": "0",
  "now": "2018-03-27 21:10:28",
  "ntpDST": "1",
  "ntpOffset": "60",
  "ntpServer": "pool.ntp.org",
  "pass0": "------------",
  "powerUnits": "0",
  "relayBoot0": "0",
  "relayBoot1": "0",
  "relayGPIO0": "12",
  "relayPulse0": "0",
  "relayPulse1": "0",
  "relaySync": "0",
  "relayTime0": "0",
  "relayTime1": "0",
  "relayType0": "0",
  "snsRead": "300",
  "snsReport": "1",
  "ssid0": "Drop it like it's hotspot",
  "telnetSTA": "0",
  "tmpCorrection": "0",
  "tmpUnits": "0",
  "tspkEnabled": "0",
  "tspkMagnitude0": "0",
  "tspkMagnitude1": "0",
  "tspkRelay0": "0",
  "tspkRelay1": "0",
  "useCSS": "0",
  "useColor": "0",
  "useGamma": "0",
  "useRGB": "0",
  "useTransitions": "0",
  "useWhite": "0",
  "wifiScan": "1",
  "wsAuth": "1"
}
```

I've hidden a couple of passwords etc with `------------` :)

lobradov, 2018-03-28:
Here's my custom.h: 

```
#if defined(LOBRADOV_BUTTON)

    // Info
    #define MANUFACTURER            "Wemos"
    #define DEVICE                  "D1 Mini + Lazar button shield"

    // Disable non-core modules
    #define ALEXA_SUPPORT           0
    #define DOMOTICZ_SUPPORT        0
    #define SENSOR_SUPPORT          0
    #define SCHEDULER_SUPPORT       0
    #define THINGSPEAK_SUPPORT      0

    #define BUTTON1_PIN             5
    #define BUTTON1_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON1_DBLCLICK        BUTTON_MODE_NONE
    #define BUTTON1_LNGCLICK        BUTTON_MODE_NONE
    #define BUTTON1_LNGLNGCLICK     BUTTON_MODE_NONE

    #define BUTTON2_PIN             4
    #define BUTTON2_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP

    #define BUTTON3_PIN             0
    #define BUTTON3_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP

    #define BUTTON4_PIN             14
    #define BUTTON4_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP

    #define BUTTON5_PIN             12
    #define BUTTON5_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP

    #define BUTTON6_PIN             13
    #define BUTTON6_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP

    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1

#endif

```

As you can see, I left most of the values to their defaults, including `BUTTON_PRESS` and `BUTTON_CLICK` and family for everything else than BUTTON1 (I disabled reset/AP/factory reset modes). 

I **think** _TOGGLE is looking for a relay that's associated to the button, and since you don't have any, it falls between the cracks.

joearkay, 2018-03-29:
@lobradov Thanks for your help so far. I've matched my config to yours, as below:

```
#elif defined(NODEMCU_LOLIN)

    // MODIFIED CODEBASE BY JRK FOR BOARD 
          #define MANUFACTURER        "NODEMCU"
          #define DEVICE              "LOLIN"

	  #define BUTTON1_PIN		  16 //D0 - Button1
	  #define BUTTON1_MODE        BUTTON_PUSHBUTTON
	  
	  #define BUTTON2_PIN         12 //D6 - Button2
	  #define BUTTON2_MODE        BUTTON_PUSHBUTTON
	   
	  #define BUTTON3_PIN         13 //D7 - PIR INPUT
	  #define BUTTON3_MODE        BUTTON_PUSHBUTTON

	  
 
	 //Relays
	  #define RELAY1_PIN          5 //D1 
	  #define RELAY1_TYPE         RELAY_TYPE_NORMAL
	  
	  #define RELAY2_PIN          4 //D2 
	  #define RELAY2_TYPE         RELAY_TYPE_NORMAL
	 
	 
	  #ifndef DHT_SUPPORT
          #define DHT_SUPPORT         1
          #endif
          #define DHT_PIN             14
```

however still no joy. Anyhow, you wouldn't expect this config to disable the mqtt messages for the switches would you? Rather, it should just change the way it reports. I'm a little lost to be honest. @xoseperez - anything obvious you can see? 

I flashed a simple Arduino sketch to the board just now to check that my buttons will all read in, which they do. This eliminates any hardware issues. 

My Device has MQTT ID `wifihomenode0` and MQTT Root Topic `wifihome/wifihomenode0`. I have subscribed to `wifihome/wifihomenode0\+` and can see all messages (uptime, temperature, humidity etc) but I cannot see any messages come through when I push my buttons :( 

Happy to issue a PR to the docs once this has been sorted.

joearkay, 2018-03-29:
I'm currently just compiling and uploading this in the Arduino IDE. I'm assuming as I'm updating the hardware.h header file I don't need to do anything with build flags to make this work? 

The only other file I edited was the `arduino.h` - Uncommented `#define NODEMCU_LOLIN`

joearkay, 2018-03-29:
I've also been reading up on a similar issue here: https://github.com/xoseperez/espurna/issues/631

We seem to be doing everything as per: https://github.com/xoseperez/espurna/wiki/Buttons-and-switches#buttons-and-switches

joearkay, 2018-03-29:
@mcspr - that's sorted it! I'm _extremely_ new to MQTT and you've saved me!

closed 2018-03-29 by joearkay