xoseperez/espurna #734: Sonoff crashes following upgrade

pcornill, 2018-03-27:
I have tried over 3 different Sonoff Basic switches. It is reproducible:

Flash a Sonoff Basic with Sonoff TH option (I use GPIO #14 and DHT22). Firmware was 1.10.
Later I upgraded with 1.12.3 (same with 1.12.4) BUT without the TH option (Back to Basic).  

It crashes after few minutes and after Power off/on, it returns to AP.
--> there is nothing coming on Console before the crash.
--> I even tried to flush the memory with esptool  erase_flash
--> Re-flashing with TH option seems to fix the problem!

xoseperez, 2018-04-03:
"There is nothing coming on Console" means nothing at all? No boot messages? It could be a bad flash.

pcornill, 2018-04-04:
No, the device works fine until the Wifi connects. 
I have Console messages up to Wifi connection and even few minutes after.
At the time of the crash, the device doesn't relate any console data. It crashes that's it!
I believe there is a link with Wifi connection because it doesn't crash in AP mode.

xoseperez, 2018-04-11:
Can you copy here the debug log when the device boots and tries to connect?

pcornill, 2018-07-02:
Xose, I believe you can close this one. I realized I used Iwip 2.0 instead of 1.4

closed 2018-07-02 by pcornill