xoseperez/espurna #735: mqtt template

tarbax, 2018-03-28:
how can i give the light a desired transition length that is not preset to espurna?

mqtt json service data:
{
"transition": "500",
"brightness_pct": "100",
"entity_id": "light.quinled_bureau"
}

mcspr, 2018-03-29:
Transition is built-time only, for now. My understanding it is planned via fastled - #320 

Ofc you can built espurna yourself and modify these values per device.
https://github.com/xoseperez/espurna/blob/8f0af3eb20c873dd620cff2a50c94de7c71ef47e/code/espurna/config/general.h#L728-L729

xoseperez, 2018-04-04:
Currently, connection with HomeAsistant is done using basic mqtt platform, not mqtt_json (the "Use JSON" toggle in MQTT tab is not related to this). This means it does not support transitions or effects.