xoseperez/espurna #737: SP1: very attractive universal multi-company smart plug with energy monitor!

knopserl, 2018-03-28:
Hi, I have found a very interesting universal smart power plug called SP1. This hase a very good rating and is available from many companies based on a base product SP1. You can find it easily in the web or on Amzon and it's offered from many companies like: Gosund, Homeplug, Coosa etc. It's always the same product and same PCB, I have opened it and soldered power and RX/TX and read the chip id: It's a ESP8266EX with 1M flash and it uses the Smart Life app and a Chinese cloud. 
It offers switch and power meter. Unfortunatly it has an energy meter chip labeled (3 lines) which you don't find in the web: HJL-01, J1749CYH, D797480E. Based on the parts around and wiring, I guess its a HLW8012. 
It would be extremly attractive to get that working with espurna (might require some reverse engineering) but I'll try it but would be nice if I could get some help to get faster to a result.
Here some links and  pictures from inside the plug:

http://www.gosund.com/index.php?m=content&c=index&a=show&catid=6&id=5
https://www.amazon.de/Steckdose-Homecube-intelligente-Verbrauchsanzeige-funktioniert/dp/B076Q2LKHG
https://www.amazon.de/intelligente-COOSA-Stromverbrauch-funktioniert-Fernsteurung/dp/B077RTJJH1/ref=sr_1_1?s=diy&ie=UTF8&qid=1522268132&sr=1-1&keywords=coosa
US version is SP2 (I guess same electronics PCB)
https://www.amazon.com/COOSA-Monitoring-Function-Campatible-Assiatant/dp/B0788W9TDR/ref=sr_1_fkmr0_3?s=hi&ie=UTF8&qid=1522267864&sr=1-3-fkmr0&keywords=coosa+SP1

[image]

[image]
[image]
[image]

gn0st1c, 2018-03-28:
that is definitely HLW8012

CURRENT_RESISTOR and VOLTAGE_RESISTOR_DOWNSTREAM values already matches with espurna's default.

you should follow the VOLTAGE_RESISTOR_UPSTREAM resistors.

knopserl, 2018-03-28:
Nice that you confirm that. I have build and uploaded a new firmware called SP1. No I have the problem, that the WiFi works, but no relay, LED and energy monitor because I don't know at all which pins are used at the ESP8266EX. There is not descripting on the board (beside 3.3V, GND, RX, TX and IO0). SO I would need endless build attempts. It would be great if the pins could be soft configured via EEPROM. I have created a new issue for that. Do you think that can work or do you have any othe hint which helps to speed up or is effective?

gn0st1c, 2018-03-29:
@knopserl : (for your pins image)
right pins top to bottom: en, 14, 13, 0, 4, RX(3), 3v3
left pins top to bottom: tout, 12, 15, 2, 5, TX, gnd

knopserl, 2018-03-29:
**@gn0st1c** : you are incrediable and saved me a lot of time! My idea whould have been to write a sketch to set/read all the ports and and measure the signal at the PCB pinout using the serial line for manual interaction. How do you know that pinout? Do you know the PCB or do you have an SP1 or what magic have you applied?
I'll now try to find out how the pins are connected to the relay, the red/blue LED, the push button and the three HLW8012 connection and will set then the proper HW pinouts in the espurna proejct. If that works, you are my hero and we have a new popular device supported by espurna!

gn0st1c, 2018-03-29:
@knopserl :
i know the pins because i've met this esp several times before :)
this may also help you; [gpio via Terminal](https://github.com/xoseperez/espurna/wiki/Terminal)

----

i'm sure you already know these but;
>**button:** check if one of the pins is connected to gnd or vcc. (there may also be a resistor for pull-up/down. this is important for definitions like BUTTON_DEFAULT_HIGH, BUTTON_SET_PULLUP
). trace the remaining pin.

>**led:** again check for gnd or vcc, trace the remaining pin. (there is most surely a current limiting resistor for the led)

>**relay:** there is surely a flyback diode in parallel to relay pins. trace it back to the transistor and transistor to esp. (there is most surely a current limiting resistor for the transistor)

knopserl, 2018-03-29:
@gn0st1c : You're totaly right with your hints. I'm a big step further, but still need some more help, so please be a bit more patient with me :-)

I think I have found out the pins and their functions.
I have updated the pins definition in hardware.h and uploaded the new firmware.
Something is now working a couple of things not.
I can remotely (via the Web)  switch on/off the relay and the LED red is on/off. But I cannot switch the relay using the physical pin. Also the power meter does not show any value (GPIO 12, 5, 14).
If I press the button long, then it does a reset and starts again the AP. So I guess the button pin is correct but I can't switch on manuall (which I think should work right?)

Here are the setting in hardware.h I'm using:
#elif defined(SP1)

    // Info
    #define MANUFACTURER        "Homecube"
    #define DEVICE              "SP1"

    // Buttons
    #define BUTTON1_PIN         13
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       2

    // Relays
    #define RELAY1_PIN          15
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            0
    #define LED1_PIN_INVERSE    2

    // HLW8012
    #ifndef HLW8012_SUPPORT
    #define HLW8012_SUPPORT     1
    #endif
    #define HLW8012_SEL_PIN     12
    #define HLW8012_CF1_PIN     14
    #define HLW8012_CF_PIN      5

    #define HLW8012_VOLTAGE_R_UP            ( 2 * 1000000 )  // Upstream voltage resistor

and here the updated pin picture
[image]

thanks a lot in advance
Heinz

gn0st1c, 2018-03-29:
it seems only setting BUTTON1_RELAY to 1 should fix your problems. (you have 1 button and one relay)

knopserl, 2018-03-29:
**@gn0st1c**  you are just fantastic! The button on/off works now and I did an OTA updload/upgrade.
Now I need your help to solve the power meter issue and I'm happy and a new espurna devive is born.
Any hints? Could it be, that the pins are still wrong numbered (so far most pins you speficed are correct), only the there SEL, CF, CF1 might not fit. I will verify a third time. I'm also 90% sure that the chip is a HLW8012 but could it be that it is another chip with 8 pins with similar schematic?

knopserl, 2018-03-29:
accidently closed the issue, sorry

gn0st1c, 2018-03-29:
@knopserl : 
just make sure you got the pins right. (right of the IC, top to bottom =  SEL, CF1, CF, GND)

also here is the data sheet for your ESP8266 (S3)
[ESP8266-S3.pdf](https://github.com/xoseperez/espurna/files/1861288/ESP8266-S3.pdf)

gn0st1c, 2018-03-29:
can you zip and attach your original firmware.bin ?

knopserl, 2018-03-29:
yes of course:
[SP1-original.zip](https://github.com/xoseperez/espurna/files/1861528/SP1-original.zip)

It's from that company
https://www.amazon.de/Steckdose-Homecube-intelligente-Verbrauchsanzeige-funktioniert/dp/B076Q2LKHG/ref=sr_1_fkmr0_1?ie=UTF8&qid=1522349887&sr=8-1-fkmr0&keywords=homecube+SP1
which is same like that
https://www.amazon.de/Intelligente-Stromverbrauch-Fernsteurung-Schaltbare-Energieklasse/dp/B076WZQS4S/ref=sr_1_1?ie=UTF8&qid=1522350030&sr=8-1&keywords=gosund+sp1
or that
https://www.amazon.de/intelligente-COOSA-Stromverbrauch-funktioniert-Fernsteurung/dp/B077RTJJH1/ref=sr_1_fkmr0_1?s=diy&ie=UTF8&qid=1522350072&sr=1-1-fkmr0&keywords=coosa+sp1

from where are you? I'm from Austria.

gn0st1c, 2018-03-29:
well, this was a very interesting decompiling session :)

the chip is BL0937. it's pin compatible with HLW8012. you learn something everyday.
luckily, we can code a new sensor driver based on the current HLW8012 sensor code.

and here is the datasheet; 
[BL0937_Datasheet_V1.01.pdf.zip](https://github.com/xoseperez/espurna/files/1861834/BL0937_Datasheet_V1.01.pdf.zip)

knopserl, 2018-03-30:
what concerns me a bit is, that the resulted values using the HLW8012 code are completly nonsens or look like it does not work at all (see the Web page screen shot above)
examples: 
current: 11A should be 0.05A
voltage: 0V should be 230V
power: 125767W should be 10W
other values are 0 or invalid.
But maybe because of the reference voltage is different the calculated values are totally invalid and not just wrong.

Yes kudos goes to @gn0st1c he is doing a great reverse engineering job and gave me the essential hints.
The switch is working but energy monitor has to be solved. I'm really looking forward and hope we get that solved.

I think it's really worth to get this device hacked/supported as it is very well rated/advertised/ranked in many articels and sold under many different brands. I have so far not found anything I like more.

gn0st1c, 2018-03-30:
i've "blindly" coded a driver but it needs testing :)

knopserl, 2018-03-30:
of course, I'm the tester, so how to I get it?

gn0st1c, 2018-03-30:
to use it in espurna, you have to add a handler as well. i'll do that later.

so just for testing purposes, you can just rename files & variables of the attached files and overwrite your current HLW lib folder. (find replace BL0937 to HLW8012)
then you can compile as before.

please note that, i just wrote the code last night and never did a second control pass. it may not work.
[BL0937.zip](https://github.com/xoseperez/espurna/files/1863593/BL0937.zip)

knopserl, 2018-03-30:
I have copied the files to the folder: 
vscode\espurna\code\.piolibdeps\HLW8012\src
renamed both to HLW8012.h and HLW8012.cpp and rebuild the project which of course gave compile errors as the HLW8012 was no longer known, so I renamed all BL0937 in the header and cpp file and ten it successfully build the new firmware. After the upload, all values are now 0.

gn0st1c, 2018-03-30:
try this. if this doesn't work, too you'll have to wait for tonight when i have the time.

[BL0937.zip](https://github.com/xoseperez/espurna/files/1863805/BL0937.zip)

knopserl, 2018-03-30:
this is the result of the latest build
[image]

knopserl, 2018-03-30:
I have now used another SP1 with the Smartlife App (China cloud) and it shows: 230.8V, 46mA, 6.2W
Is there anything we can debug to come further? How can I support you?

gn0st1c, 2018-03-30:
no worries, we'll make it work. but later tonight. :)

gn0st1c, 2018-03-30:
i found a wifi plug with hjl markings.
i'll be flashing and testing tomorrow. it's better than the guess work.

knopserl, 2018-03-30:
ok, good luck, I'll be watching this thread. 
BTW, I have found another flaw, but guess has nothing to do with this device:
When the modem changes the IP adress, the plug seem to get and stay disconnected. It does not reconnect to WiFi (need to replug it to mains) thene it works again.

gn0st1c, 2018-03-31:
good news, i got it working.

with my HJL-01 device the SEL, CF and CF1 had pull up resistors.  it looks like your device has too. r17,r18,r20 & r15, r19, r21 perhaps?, can you check?

if so, my code has this and i'll leave it there for the time being.
pinMode(_cf, INPUT_PULLUP);
pinMode(_cf1, INPUT_PULLUP);
_attach(this, _cf, FALLING);
_attach(this, _cf1, FALLING);

knopserl, 2018-03-31:
hello **gn0st1c** great job again.
sorry I was out the last couple of hours.

that is very good news, as you had a device with the same engery meter chip and I think also the same schematic.
regarding the pull up, yes I have the same resistors, please look at the picture I have posted above and I have copied it again here. Check if you have the same values.
What do I need to do now? Where is the code to change in the HLW8012.cpp file?

[image]

gn0st1c, 2018-03-31:
@knopserl : those codes goes somewhere else.

i don't like spending too much time in interrupts so i'll clean up my code and PR as soon as possible.

---
is this how they are connected?
(10 and 10k voltage divider which makes almost no difference)

[image]

knopserl, 2018-03-31:
**gn0st1c** : exactly as you described it. Wiring and resistor values (10 and 10k). Waiting for your code and instructions.

knopserl, 2018-04-03:
**@gn0st1c** : If you don't mind and the code cleanup PR takes some time, it would be nice if you could send me your intermediate version of HJL-01 driver file?

swilson86, 2018-04-04:
This SP1 is also available with an American style outlet and seems to be similar in basic functionality to the American style ITead S31 (which I have not seen in a European version yet). Jose is working as I write this on the S31.  Thanks to all of you, knopserl, gn0st1c, and Jose for all the days that you give us that make me feel like I am a child in a candy store... I feel your energy!!

knopserl, 2018-04-04:
@swilson86 you're welcome. 
I think the SP1 support will be a big improvement of this project as it's availble in many brands and seem to be and become more and more common. I also have the ITEAD S31 on my radar but we need a European version. I have already contacted ITEAD as there are many requests for an EU version. On the other hand, if the SP1 support is finalized - thanks for gn0st1c great support - you might not need the S31 or the EU  version as the SP1 can do the same and is very cheap (14€).
I waiting now for the completing of the energy chip driver from gn0st1c then I can use my 4 pieces of SP1 (different brands, ame HW)

gn0st1c, 2018-04-05:
sorry for the belated reply.

1) last weekend, my testing device died with a bang (bigclive would be proud). so i ordered a new one, along with s31 and couple of new plugs & esp devices.

2) instead of a stand alone library, i thought it would be better if a single library handled alternative chips. thus, i changed my code so that @xoseperez 's HLW8012 library would work for HLW8012, CSE7759 and BL0937.

if you don't like to wait, just give me your mail address, i can send you the library and you can test it.

knopserl, 2018-04-05:
@gn0st1c : Thanks for your reply and that you have not forgot me :-) Just kidding but I really appreciate your great work but also can't wait to get your solution for testing.
My mail adress is knopserl-at-yahoo.de I would be happy to do some testing meanwhile.
I think your idea makes sense if you combine multiple energy monitor chips to one sensor library if most of the code can be shared and controlled by one or a few build parameter.

gn0st1c, 2018-04-08:
_just to inform those following this issue;_
yesterday, thanks to @knopserl being patient with me and sending several debug outputs, we managed to get the driver work as excepted.

i'm now waiting for my new wifi plug to arrive so i can connect it to my oscilloscope and verify some of my findings. then we'll release the driver into the wild :)

knopserl, 2018-04-24:
the driver is in beta testing and works in principle.
We have to be patient and wait. It will hopefully included in one of the next versions. gn0st1c is working on it and need some more time.

gn0st1c, 2018-04-25:
@stif: i'm on medical leave and i'm supposed to stay away from things emit light.
i'm getting better and hopefully finish the driver soon.

HobbytronicsPK, 2018-05-14:
@gn0st1c May you please share your modified library? I have Sonoff Pow with cse7759 chip inside? Kindly email me at ahmed [at] hobbytronics.pk

ColinShorts, 2018-05-14:
@HobbytronicsPK have you tried the SONOFF_POW_R2 firmware, it has a CSE7766 afaik... Check issue #761

gn0st1c, 2018-05-14:
@HobbytronicsPK : there is no standalone library for cse7766.
but you can [use this code](https://github.com/xoseperez/espurna/blob/dev/code/espurna/sensors/CSE7766Sensor.h) to develop your own.

knopserl, 2018-05-14:
aaahh, the thread I've opened gets more and more interest. The COOSA SP1 is one of the plenty of companies (beside Gosund, Homecube etc.) who is selling SP1 derivates and yes it has the BL0937 chip which is labeled with HJL-01 :-)
Thanks to gn0st1c who "reverse engineered" the chip and updated the HLW8012 driver to support this energy monitor. But is still in beta.

HobbytronicsPK, 2018-05-16:
@gn0st1c Have no clue, how to use it. If you have worked with it already, Can you please describe, how to use it?

stif, 2018-05-28:
Hi @gn0st1c, hope you are feeling better now. Do you have any news about the driver? I am happy to do some beta testing if you send me your code to stif[at]gmx.at

doodah33, 2018-06-10:
+ 1 would love to see this driver released into the wild, even if it is in beta form. I'm seeing more and more of these pop up at great prices. Using Espurna is way far more preferable to using the bundled app.

Couple more US examples using this chip:

https://www.amazon.com/Smart-Plug-Wifi-Mini-VANZAVANZU/dp/B078PHD6S5
https://www.amazon.com/Compatible-Assistant-Monitor-Required-Function/dp/B077R18JXG

xoseperez, 2018-06-11:
@gn0st1c You might want to release the driver as it is now. We could help debugging it...

knopserl, 2018-06-17:
@gn0st1c and @xoseperez I'm testing it already since I got the library and I would say it works so far and should be tested by a larger test community. I would also support that it goes to the dev branch.

gn0st1c, 2018-06-20:
i haven't noticed the latest messages. thanks to @knopserl letting me know.

@knopserl uses the driver on his 7 plugs so i'll merge it.

doodah33, 2018-06-20:
Woo hoo! Thanks, @gn0st1c

xoseperez, 2018-06-20:
HI
I received a Blitzwolf BW-SHP2 (https://www.blitzwolf.com/Wifi-Smart-Socket-EU-p-244.html) yesterday. It has the very same board inside with an HJL01 power monitoring chip. While waiting for @gn0st1c PR to the HLW8012 library to support both I managed to make it work with the same library without modifications, only small changes in how ESPurna configures it.

As @lobradov pointed out, the HJL01/BL0937 has different configuration parameters but I also found out the SEL pin is inverted (compared to the HLW8012 boards I have checked so far) and the interrupt won't work unless it monitors FALLING edges. Fortunately, all these changes can be configured from ESPurna without the need to changed the library.

I have committed my changes to the sensor branch, will join them to dev (and master) as soon as I have some feedback from you.

Of course, once @gn0st1c pushes his PR that would be the preferred way to implement it.

doodah33, 2018-06-21:
Well would you look at that

[image]

This is a 75 watt light bulb

BTW, my plugs have the following pin config:

```
    #define HLW8012_SEL_PIN     3
    #define HLW8012_CF1_PIN     14
    #define HLW8012_CF_PIN      5
```

xoseperez, 2018-06-21:
@doodah33 What device is yours? A Vanzavanzu (https://www.amazon.com/Smart-Plug-Wifi-Mini-VANZAVANZU/dp/B078PHD6S5)?

Camposmansi, 2018-06-21:
xoseperez eres un crack, el puto amo jajaja

Anxious I am waiting for the bin

doodah33, 2018-06-22:
@xoseperez that is exactly it. There are a ton that look like that one, but I couldn't vouch that they have the same components inside.

Camposmansi, 2018-06-22:
Pido perdon por mi ingles
Es exactamente el mismo, pero por lo que entiendo estais todavia investigando el tema, estoy ansioso por el bin

I apologize for my English
It's the same, but that's why I understand you're still investigating the subject, I'm looking forward to the bin

xoseperez, 2018-06-22:
Released with 1.13.0. Closing.

closed 2018-06-22 by xoseperez

knopserl, 2018-06-23:
a great and successfull thread!

nightfly0815, 2018-06-25:
You have to use the blitzwolf.bin 
I also flashed this sp1 socket and it worked!
https://www.amazon.de/Intelligente-Steckdose-Ablue-Funktioniert-Assistant/dp/B076DRFRZC/ref=sr_1_5?s=diy&ie=UTF8&qid=1529908498&sr=1-5&keywords=wlan+steckdose+smart+life
The Voltage and the Power seams to be ok.
I just have to check the ampere, with a multimeter and calculate the values. 
I´m not 100% sure if the ampere are correct shown. I saw yesterday that the decimal place was not correct. In the webif I shows 0.035A, but I think it should 0,35A. I will measure and calculate it, when I have a 100% resistive consumer.
I will give a feedback what I found out.

regards from Austria ;-)

Camposmansi, 2018-06-25:
Oohhh gracias, y no lo vi, buscaba algo mas comun como SP1

Oohhh thanks, and I did not see it, I was looking for something more common like SP1

kuppe234, 2019-09-13:
> Hi guys, do anyone know what is the power supply ic used on these
> smartplugs????

Hello,

there is an ic in SOT-23(5) package. He is named "IADGH". But I don´t
find a datasheet.
I made 2 pictures and a (very) simple schematic. You found them at the
attachment.

Greeting.

gn0st1c, 2019-09-13:
@kuppe234 : 
[mp150 web](https://www.monolithicpower.com/en/mp150.html)
[mp150 pdf](https://media.monolithicpower.com/document/MP150_r1.14.pdf)

cjheath, 2019-09-24:
Some folk mentioned they had difficulty finding the manufacturer and data sheet for the BL0937. The manufacturer is Shanghai Belling Co, Ltd. and the data sheet is available on thejir web site here: [http://www.belling.com.cn/en/product_info.html?id=138](http://www.belling.com.cn/en/product_info.html?id=138)

[7 more replies]