xoseperez/espurna #738: HW Pins configured by SW and stored in EEPROM

knopserl, 2018-03-28:
I'm trying to create support for a new device, the SP1. I have created a new Issue for that and hope others can help to get faster to a success.
I have tried to start with something which has similar features like the KMC plug with energy monitor.
Flash worked and also WiFi. Thus I can configure all via the web page, but relay, LED, energy monitor does not work. I guess the pinout don't match. 
Now you can imagine how many combination I need to try with build flash and fail.
It would be very nice if the switch, LED, energy monitor could be soft-configured (I think in the tasmota project, this could be done). Then I just need to confige it via the Web-Page vial trial/error.
Would that be possible? That would be great!

gn0st1c, 2018-03-29:
gpio command via [Terminal](https://github.com/xoseperez/espurna/wiki/Terminal) may help.

but it's always easier to track the pcb.

knopserl, 2018-03-29:
thanks for the hint. That would help of course to do some testing without the need to write an own sketch. Meanwhile with your help we figured out the pinout of the ESP8266-S3 module. Withouth the pinout of the MCU (hidden in the small metal case) it's quite some effort.

However would it not be a good idea to have the pin assignment in the EEPROM, read during startup, configured via the web UI?

xoseperez, 2018-04-03:
Yes, it's something I want to do. That is the main feature for version 2.0.

closed 2018-04-04 by xoseperez