xoseperez/espurna #739: 801 RGBWW Controller fadeing speed

bigd-73, 2018-03-29:
hi,

where can i change the fadeing speed for the 801 rgbww controller if i switched the light on or off?

regards
mario

Sanfam, 2018-03-30:
@bigd-73 , there are two constants you should be able to override that govern transition speed.  I believe both of these are configured in the compiled code and are found within [code/espurna/config/general.h](https://github.com/xoseperez/espurna/blob/master/code/espurna/config/general.h#L703):
```
#define LIGHT_TRANSITION_STEP   10          // Time in millis between each transtion step
#define LIGHT_TRANSITION_STEPS  50          // Number of steps to acomplish transition
```

Reducing the number of milliseconds between steps or reducing the number of steps should speed up each transition, while increasing these values should slow it down.

xoseperez, 2018-04-03:
Staged for release

closed 2018-04-03 by xoseperez

bigd-73, 2018-04-09:
Hallo!

Thank you for integrating the feature in version 1.12.5. But where in the WebUI can I find the functions?

Regard