xoseperez/espurna #740: RGBWW 801 Controller light is flickering

bigd-73, 2018-03-29:
hi,

sometimes the rgbww 801 controller is flickering. is this a software or hardware problem?

regards
mario

Sanfam, 2018-03-30:
Could you clarify what you mean by "Flickering?"  This could be interpreted as the lights briefly flashing off-and-on again, the lights strobing at a specific frequency (ex. 30, 40, or 60 hz), or something else entirely.

flashy02, 2018-04-04:
Hi, I have this problem too. While turning on/off (fading) the light is flickering. 

regards 
Michael

flashy02, 2018-04-10:
Hi, then it's an other problem. My controller works fine. Yesterday I switched from 1.12.3 to 1.12.4, so I will verify the behaviour. But actual it still works.
Best regards

xoseperez, 2018-04-10:
@bigd-73 What version are you using? Some development version had such behavior (it is actually resetting) over the last month (pre-1.12.5). Try updating to the last released version.

mrcodegh, 2018-06-01:
I had also seen this problem in version 1.12.3, however in recent version 1.12.6 it appears to be fixed (30+ hours observation).  Version 1.9.7 was also free from brief flickers every ~couple hours so bug was introduced after 1.9.7 and likely corrected before 1.12.5.

xoseperez, 2018-06-04:
@bigd-73, can you verify @mrcodegh words?

bigd-73, 2018-06-05:
I can't test version 1.12.6. I don't have a stable wireless connection with this version.

[https://github.com/xoseperez/espurna/issues/914]

closed 2018-08-20 by stale[bot]

Adrianotiger, 2019-01-22:
Sorry if I write it on this closed issue.
I have the same problem with the H801. It flicker (lights up to maximal power for a moment) when it is fading from a low value.
It doesn't happens, if all channels are fading with the same value or if only 1 channel is fading.
It happens if there are 4 channels fading with different values.
If you have this situation, the last fading step (maybe with value 5 or 10) will bright up your light.

It has nothing to do with this code I thing, because the values are always set correctly.

As workaround, you can add this code in the light.ino file, in the _toPWM function:

```
unsigned int _toPWM(unsigned long value, bool gamma, bool reverse) {
    value = constrain(value, 0, LIGHT_MAX_VALUE);
    if (gamma) value = _light_gamma_table[value];
    if (LIGHT_MAX_VALUE != LIGHT_LIMIT_PWM) value = map(value, 0, LIGHT_MAX_VALUE, 0, LIGHT_LIMIT_PWM);
    if (reverse) value = LIGHT_LIMIT_PWM - value;
    // WORKAROUND (H801):
    if(value < LIGHT_MAX_VALUE / 50) 
      value = value > LIGHT_MAX_VALUE / 100 ? LIGHT_MAX_VALUE / 50 : 0;
    return value;
}
```
It solved my problem, using a LIGHT_MAX_VALUE of 4095.