xoseperez/espurna #744: Cannot change state via MQTT

wsw70, 2018-03-31:
Hello

I have flashed ESPurna on a Sonoff Basic and tried to change its state via MQTT by sending the following message to the MQTT broker :

`ESPURNA_A3AD46/relay/0 1`

There is no reaction of the Sonoff device. When pushing its button, I get message form it:

```
ESPURNA_A3AD46/button/0 1
ESPURNA_A3AD46/button/0 2
ESPURNA_A3AD46/relay/0 1
ESPURNA_A3AD46/button/0 1
ESPURNA_A3AD46/button/0 2
ESPURNA_A3AD46/relay/0 0
```

(the button was pushed twice, to switch it on an off).

Is there anything particular to do to to make it listen to the state topics, in addition of the sending (which works)?

xoseperez, 2018-03-31:
The command topics end with `/set` by default. So you have to send it to `ESPURNA_A3AD46/relay/0/set`.

wsw70, 2018-04-01:
Thank you so much. I looked though the documentation but somehow did not find it - is it somewhere documented?
I am asking as I would be glad to update the Wiki (which is really great) and add some elements if you are fine with that (I already updated a minor point with the initial access to the built-in AP)

closed 2018-04-01 by wsw70