xoseperez/espurna #746: H801: Self compiled version just die after 10 second uptime

Skaronator, 2018-04-02:
Hey,

I try to compile espurna with Arduino IDE for my **H801**. I'm using the **MASTER**-Branch and it compile file and also the flashing works fine but suddenly it just die (and sometimes reboot). I've set my Wifi Data in the `custom.h` file and it connects to my AP. So I'm running a Ping command and after ~10 second it just lost the connection.

```
C:\Users\Skaronator>ping 192.168.178.201 -t

Ping wird ausgeführt für 192.168.178.201 mit 32 Bytes Daten:
Zeitüberschreitung der Anforderung. (plug in device)
Antwort von 192.168.178.201: Bytes=32 Zeit=6ms TTL=255
Antwort von 192.168.178.201: Bytes=32 Zeit=3ms TTL=255
Antwort von 192.168.178.201: Bytes=32 Zeit=6ms TTL=255
Antwort von 192.168.178.201: Bytes=32 Zeit=1ms TTL=255
Antwort von 192.168.178.201: Bytes=32 Zeit=4ms TTL=255
Antwort von 192.168.178.201: Bytes=32 Zeit=1ms TTL=255
Antwort von 192.168.178.201: Bytes=32 Zeit=1ms TTL=255
Antwort von 192.168.178.201: Bytes=32 Zeit=1ms TTL=255
Antwort von 192.168.178.201: Bytes=32 Zeit=1ms TTL=255
Antwort von 192.168.178.201: Bytes=32 Zeit=10ms TTL=255
Zeitüberschreitung der Anforderung. (Timeout)
Zeitüberschreitung der Anforderung. (Timeout)
```

Here some infos:
Using Adruino 1.8.5 and installed the esp8266 2.4.1 version.

All required plugins (default values):
- ArduinoJson 
- Embedis 
- AsyncMqttClient 
- DebounceEvent 
- ESPAsyncTCP (latest)
- ESPAsyncWebServer  (the old version with the linked commit)
- FauxmoESP 
- JustWifi
- NtpCLientLib
- Time

And here a screenshot of Arduino

[image]

My custom.h looks like this:
```
#undef DEVICE_NAME
#undef WIFI1_IP
#define DEVICE_NAME "H801-201"
#define WIFI1_IP "192.168.178.201"

#undef ADMIN_PASS
#undef WEB_FORCE_PASS_CHANGE
#undef WIFI_SCAN_NETWORKS
#define ADMIN_PASS "S7ROr6VLhiEKL5QsD5AQIuTMQv3R1H2"
#define WEB_FORCE_PASS_CHANGE 0
#define WIFI_SCAN_NETWORKS 0

#undef WIFI1_SSID
#define WIFI1_SSID "FritzBox"
#undef WIFI1_PASS
#define WIFI1_PASS "z3fcY9xyTJuwGjHAfXbxBYNy"
#undef WIFI1_GW
#define WIFI1_GW "192.168.178.1"
#undef WIFI1_DNS
#define WIFI1_DNS "192.168.178.4"
#undef WIFI1_MASK
#define WIFI1_MASK "255.255.255.0"
```

I've move the `include` in `all.h` below the `sensor.h` include. Thats why the `#undef`s

Do you have any clue whats wrong?

Skaronator, 2018-04-02:
Everything is fine when using the pre-compiled version so something is wrong with my compiler.

xoseperez, 2018-04-02:
Try using lwIP 1.4 instead of 2.0

Skaronator, 2018-04-02:
Same issue after switching to lwIP 1.4. I also switched to dev branch and tested it again as well but same result.

I've downgraded the ESP8266 Arduino Core to 2.3.0 and got the same result for dev and master branch. I can't change the lwIP anymore since downgrading the Core version.. mh.

[image]

Skaronator, 2018-04-03:
Looks like I got it working. I started from 0 again and seems like 1-2 library versions were wrong.

Also found a typo in the Wiki page:
"If you are using Arduino core 2.3.0 stable, please install and use this version https://github.com/me-no-dev/ESPAsyncTCP#9b0cc37"
the link doesn't correctly work. This should be the correct link:

https://github.com/me-no-dev/ESPAsyncTCP/tree/9b0cc37cd7bdf4b7d17a363141d2e988aa46652c

I've updated the page.

closed 2018-04-03 by Skaronator

xoseperez, 2018-04-04:
Thank you @Skaronator !