xoseperez/espurna #750: Add Telegram notification support

mstjerna, 2018-04-04:
Hi

I would love to see a Telegram bot implemented.
It would be great to schedule a notification if the temperature rices above a certain level or a GPIO is changed (PIR motion for example).

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot

lobradov, 2018-04-04:
Isn't that a role of home automation logic (home assistant, domoticz, node red or similar software)?

mstjerna, 2018-04-04:
Sure thing, but if you only have a basic sensor it would be if it could send messages. 
I use the OpenHAB for Telegram at home but haven't got around installing it at my parents place or in my summerhouse...

xoseperez, 2018-04-11:
This issue raises several questions. First, you will not only need the telegram bot implemented but a alert system (you said "if the temperature rises above..."). This is not implemented in the firmware. Because of the second point.

And second, in the line of what @lobradov says, these kind of features are better implemented outside the device because services like Node-RED are far more powerful and flexible and it's incredibly easy to gather different infos a build your own intelligent home system.

Now, I understand that if you are only going to have one sensor or you don't want to mess with other stuff you'd like to have these kind of features in the device, so it is as autonomous as possible. Problem is that there are a zillion different possibilities out there and very limited time so there should be a set of reasons to prioritize one feature or another.

* Community impact is one: if it is a feature that will be used by a large number of people it makes sense to implement it. The scheduler fits into this category.
* Versatility: sometimes a feature is an enabler for other things, MQTT for instance.
* Contribution. Well, if someone has work on something for the community I will not say no, at most the feature might not be enabled by default.
* Coolness. Of course.

Having a Telegram bot would be cool and probably a good feature for the community. But point number one above is still pending...

Camposmansi, 2018-09-26:
I join

jjavier210, 2021-02-05:
i join :D