xoseperez/espurna #751: How to remotely set the hostname?

wsw70, 2018-04-04:
I deploy a custom-built firmware to my devices and perform a factory reset after the deployment so that all parameters are set to the custom settings in my firmware (I patched `web.ino` to add a  `/factoryreset` endpoint which mimics the `/reset` one but with `CUSTOM_RESET_FACTORY`).

This has the side effect to also set the default hostname to `ESPURNA_SOMETHING`, overwriting the only localized info of the firmware (= the only information which is "hand setup" for each device).

**Is there a way to remotely set the hostname of a device?** (via a HTTP call)

mcspr, 2018-04-04:
I am pretty sure that uploading previous backup and sending to the `/config` also does this. You could modify it and only leave `hostname` and `app` (required by `settingsRestoreJson()`). Everything will be erased but the uploaded keys.
```
$ cat espurna.json
{"app": "ESPURNA","hostname": "ANRUPSE_123456","ssid0": "WiFi","pass0": "12345678"}
$ curl -F "file=@espurna.json" -u admin:fibonacci http://10.11.12.13
$ curl -u admin:fibonacci http://10.11.12.13/config
{
  "app": "ESPURNA",
  "version": "1.12.5b",
  "hostname": "ANRUPSE_123456",
  "pass0": "12345678",
  "ssid0": "WiFi"
}
```

wsw70, 2018-04-05:
@mcspr thank you - this is a very good idea. 

I pushed remotely a minimal configuration (`hostname`, `app`,  `version`and `mqttTopic`) to try that out  but it bricked my device. This is not a problem, I will reflash it tonight when back home, my wild guess is that the version I pushed had incorrect content (some characters, probably, or the length)

```
{
  'app': 'ESPURNA',
  'version': '1.12.5b-2018-04-05T12:29:05.828801+02:00',
  'hostname': 'maison-relay1',
  'mqttTopic': 'maison/relay1'
}
```

closed 2018-04-05 by wsw70