xoseperez/espurna #755: MQTT: Brightness calculation is currently wrong in master-branch (and probably dev as well)

Skaronator, 2018-04-04:
Something is wrong with the brightness calculation. I can reproduce this 100% in Home Assistant via MQTT.

1. Change brightness to a other value. (100% -> 50%)
2. Select a other color
3. Change brighness back to original value (50% -> 100%)
4. Click on **the same*** color and you see that the brighness actually change (which shouldn't be the case)

* You don't have to select the same color but it is easier to see the brighness difference.

My HA Config:
```
light:
  - name: H801_200
    platform: mqtt
    state_topic: H801-200/relay/0
    command_topic: H801-200/relay/0/set
    payload_on: 1
    payload_off: 0
    availability_topic: H801-200/status
    payload_available: 1
    payload_not_available: 0
    brightness_state_topic: H801-200/brightness
    brightness_command_topic: H801-200/brightness/set
    rgb_state_topic: H801-200/rgb
    rgb_command_topic: H801-200/rgb/set
    color_temp_state_topic: H801-200/mired
    color_temp_command_topic: H801-200/mired/set   
```

Skaronator, 2018-04-06:
I might found the reason for this bug. This code updates the `_light_brightness` variable:

https://github.com/xoseperez/espurna/blob/fa20c47f2905c46628cbdde475e899ecc30368a3/code/espurna/light.ino#L529-L533

which later update the RGB values and apply the brightness:

https://github.com/xoseperez/espurna/blob/fa20c47f2905c46628cbdde475e899ecc30368a3/code/espurna/light.ino#L387-L390

But the RGB values (`_light_channel[i].value`) in this calculation have the brightness already applied.

Possible fix: Move the brighness setting to a new variable (`_light_channel[i].brighness`) and apply the brighness in the very last step.

Skaronator, 2018-04-06:
I did some debugging and it looks like it is Home Assistants fault?

https://github.com/home-assistant/home-assistant/blob/3394916a68122913c7b066b50d217cfe298914eb/homeassistant/components/light/mqtt.py#L427-L428

closed 2018-06-21 by Skaronator

1043717432, 2019-03-16:
> 在Home Assistant中修复

How do you deal with success in the end?