xoseperez/espurna #764: Plugin support

nunofgs, 2018-04-06:
Hi Xose, thanks for all your work. I've been using this firmware since the early versions and its incredible how far you've taken it!

Anyway, do you have any plans to add support for user-developed plugins? The reason is I have a very specific integration in mind that would most likely be rejected as a PR (since it's so specific).

xoseperez, 2018-04-08:
You have several "tools" to do so. There is the custom.h file to add custom configuration settings, the USE_EXTRA setting to force calling a certain setup method for your plugin and several callback register methods to bind your code to the main loop, MQTT, WIFI or the local broker (and get relay and sensor notifications). 

Of course this might not be enough. But maybe you can start working on this and we can merge other integration tools you feel necessary.

zafrirron, 2018-04-09:
Hi @nunofgs 

Please see the attached files, were I've created a plugin template/tool/model (all files includes inline help documentation for all possible options).
 
## Files:
* **custom.h** - this is the generic espurna custom header include, included by all.h based on USE_CUSTOM_H flag and includes plugin(s) definitions (an activation flag, plugin header file include and plugin setup function call.

* **plugin1.h** - The plugin header file (should be placed in config folder) includes plugin specific defines. this header is included by custom.h based on internal flag (PLUGIN_1) to allow switching on/off multiple plugins.

* **plugin.ino** - The plugin template, to be placed in code folder. this template enables writing a plugin and use all espurna services and utils (I may missed some....but will improve).

## Status:
In development phase, working, partially tested.

## Limitations 
Due to current frontend structure, I did not add the HTML/JS/CSS files, since I did not want to change core files, so in case of frontend need for plugin, you need to add the plugin page, but all the web services hooks included in the template (dont forget to gulp)
 
## Template structure:
(All inline documented )
Like any other espurna module, includes helper functions (your plugins magic code), setup part and loop part.
setup function (pluginxSetup()) called by extraSetup() defined in custom.h and called by espura.ino setup function based on USE_EXTRA flag (this is the generic espurna 3rd party integration hook).
plugin main loop called by espurna (plugin loop is registered by pluginsetup function)

## Template features:
(services enabled by the plugin template), most code is commented out, to allow uncomment/delete needed/unnedded code. The services are all public espurna functions to be used for the specific plugin needs logic and data.  

* Main loop features
   * Get sensors data
   * Set relays
   * Factory reset
   * AP mode
   * System reset
   * Setter/Getter for plugin params
   * Broker publish 
   * MQTT publish
   * DOMOTICZ publish
   * INFLUXDB publish
   * THINGSPEAK publish
 For demonstration only the main loop currently counting and debug reporting every 5 sec a loop counter.

* Setup features
  * HTTP API registration - currently activated for demonstration
  * Web services 
  * Terminal support - currently activated for demonstration

## Please feel free to give any feedback/comment/suggestions  
 
[Espurna_Plugin.zip](https://github.com/xoseperez/espurna/files/1889850/Espurna_Plugin.zip)

zafrirron, 2018-04-12:
@nunofgs 
thanks for your feedback, I'm not sure if I understood you correctly, my plan was to enable using latest espurna version to enable creation of your personal version including your plugin.
The plugin should include all your unique definitions (overriding defines and so... ) to become independent of new version.
Since the loading is single image loading, I guess you meant activating/deactivating of plugin, this is definitely a good idea!
another Item I'm working on  is adding a generic hook system so inside any espurna module it will be possible to create a hook and in any plugin possible to register to this hook.

Currently my template enables to run a private ("3rd party") code the can use any espurna feature but does not change espurna core behaviour. 
I real plugin system also register to hooks  and change/add functionality to core. for example if you want to run a code before espurna changes a relay status, you just register to 'RELAY_BEFORE' action hook with your callback function (assuming core espurna code include this hook as Xose mentioned).

zafrirron, 2018-04-13:
Ok,

See my new pull request regarding generic plugin hooks, exactly as you described, a very simple code enables registration of hooks.
Regarding separate folder, correct I missed that one. 
all plugins should be under separate folder (I'll fix it in the template).
Regarding UI, I'm looking into a UI templating model, I saw some gulp implementations that include templating system so the front end could be  templated, this will also allow plugins UI.
my view on the architecture is that some of the very hard work you perform on integrating new sponsors and  hardware, could be moved to plugin model, and the main focus of the core development should be the backbone of core system.
Sensors and new boards should be easy to add with out core changes.

zafrirron, 2018-04-13:
Regarding separate plugin folder, it may be my frameworks knowledge limitations, but after reading PIO docs, forums and testing, it looks like the plugin header files could be moved to seperate folder.
(PIO lib model refer to "/lib/plugin1" folder structure were you should put library code into)
trying to put plugin code into same folder however creates a little mess since PIO knows to link lib folder files if header included from .ino files on "src_dir" folder.
so for now I moved only the plugin headers to /lib/plugin(x) folder and will look into possible options (maybe one of the frameworks gurus will guide me...)

xoseperez, 2018-04-13:
You mention it. I'm pretty sure it will work with PlatformIO but, again, 80% (to say a number) of ESPurna users use Arduino IDE which is far more limited, maybe not when it comes to build the firmware, but I'm afraid it does not support autoloading files (in tabs) in a recursive way. For newbies this would mean they will have to know these files exist and load them files manually and this add complexity...

uberflyx, 2018-04-13:
Hi all. This plugin idea is great. I'm thinking of adding some extra sensors as plugins rather than, well, sensors. I have some thoughts: on the latest code, to use a plugin one has to include custom.h. I don't think this is the way to go as it confounds the point of custom.h. I think it will be better to create a plugins.h file which is included by all.h, and then the "user" can uncomment the plugins they wish in the plugins.h file. Otherwise, the architecture seems like it will work nicely.

zafrirron, 2018-04-13:
Agree about the entry point, I had to select it for existing integration, custom.h was the only entry point.
A more suitable way will be plugin.h (I'm working on improved structure for it), and change the USE_EXTRA to control the plugin.h inclusion (USE_PLUGIN?) this is existing integration point.

@uberflyx if you have some thoughts about folder structure for the plugins that will fly for all development frameworks, please share....(header files looks easy but plugin c code files looks little more trickier).

Misiu, 2018-04-24:
I'm using ESPurna as an awesome code base, having plugin system and scheduler would allow me to drop my custom code and focus on supporting ESPurna.
I need simple temperature logging to SD card every X minutes.
Plugin system would allow me to create simple 2 files instead of hacking whole source code.

GULP is already used so maybe it can be reused when creating Web UI for specific plugins?
In my case I need to be able to specify minutes between writes to SD and be able to download file that contains data.

Looking forward for plugin support 👍

zafrirron, 2018-04-24:
@Misiu 
First iteration (MVP?) for plugin option described here..
https://github.com/xoseperez/espurna/wiki/3rd-Party-Plugins

This very first version doesn't include UI yet, but you could try to use it and have your parameters injected thru API (included in the template) or terminal command...

I'm working on next version, that will include web support and some other features, but it will take some time...

gn0st1c, 2018-04-25:
@zafrirron: how about using gulp-inject?

you can have placeholders for plugin's html/css/js codes and if the plugin folder has any of the file(s), they will be injected. if there is no file, it will be skipped and remarks will be removed anyways.

https://www.npmjs.com/package/gulp-inject#injecting-files-contents

```
inside index.html:
  <!-- inject:espurna/plugins/plugin1/plugin_menu.html -->
  <!-- endinject -->
  <!-- inject:espurna/plugins/plugin2/plugin_menu.html -->
  <!-- endinject -->

  <!-- inject:espurna/plugins/plugin1/plugin_panel.html -->
  <!-- endinject -->
  <!-- inject:espurna/plugins/plugin2/plugin_panel.html -->
  <!-- endinject -->

inside custom.css:
  <!-- inject:espurna/plugins/plugin1/plugin.css -->
  <!-- endinject -->
  <!-- inject:espurna/plugins/plugin2/plugin.css -->
  <!-- endinject -->

custom.js:
  <!-- inject:espurna/plugins/plugin1/plugin.js -->
  <!-- endinject -->
  <!-- inject:espurna/plugins/plugin2/plugin.js -->
  <!-- endinject -->
```

template UI:
[plugin_UI.zip](https://github.com/xoseperez/espurna/files/1948179/plugin_UI.zip)

things to consider:
* a bad html/css/js may screw up the whole UI
* tabindex

gn0st1c, 2018-04-25:
@zafrirron : there were some minor typos inside plugin_0.0.1.zip's ino file. if you'd like to merge to yours;
[plugin1.ino.zip](https://github.com/xoseperez/espurna/files/1948231/plugin1.ino.zip)

zafrirron, 2018-04-25:
@nunofgs 
Thanks!
The UI solution sounds like a great option...
I will try it later today and add it to the wiki page...

mcspr, 2018-04-26:
@gn0st1c Wouldn't htmllint part of gulp chain cover that? It did break build for me with invalid tags (not closed, slash position etc.) altough it was not really helpful showing actual error.
Similar linter for resulting js could be added. And tabindex could be generated on panel load by js itself.

@zafrirron What did you try? It is not required to use node though, my personal preference would be python+jinja2. For example, platformio uses SCons.Scanner - see [platformio-core/builder/tools/piolib.py](https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/piolib.py) - and using that data extra_script could render resulting html
via pre-action on espurna.ino.cpp.o target: *extra_script called -> render index.html -> call gulp -> package .gz.h -> header is packaged inside .o*

lblabr, 2019-01-13:
is there some progress in here ?