xoseperez/espurna #766: Question about Sonoff TH

paveleremin, 2018-04-07:
Any body know what should be done to get dry contacts instead of main voltage at Sonoff TH?

paveleremin, 2018-04-09:
I found only this information:
[image]
Here is my implementation:
[image]
but seems still mains at the output
@xoseperez maybe you know?

gn0st1c, 2018-04-09:
description looks good, implementation not so much. (looks like there is solder residue around the relay contact and the Live In fuse track). Make sure there is enough gap in between. 

[image]

gn0st1c, 2018-04-09:
@paveleremin : my suggestion would be just desolder the relay. cut the tracks. resolder relay back.

closed 2018-08-17 by stale[bot]