xoseperez/espurna #767: Can the smartconfig function be implemented？

aniceking, 2018-04-08:
WiFi.beginSmartConfig(); 
WiFi.smartConfigDone(); 

Where can these two functions be added, and they will be useful?

zafrirron, 2018-04-08:
Second this!
This will save lots of time configuring multiple devices, eliminating the need to configure each device wifi configuration.
To add (little) security, this could be done like WPS setup, enabling "discovery" will start the smartconfig mode and allow other devices to be auto configure their wifi, then smart config will be terminated.

mcspr, 2018-04-08:
Interesting feature. If i understand correctly - smartConfig() creates AP after unsuccessful connection attempt and external smartConfig client (phone) connects to it and sends udp data about wifi connection. 

ESPurna already creates `<devicename>` AP and it supports configuration upload via `http://192.168.4.1.lan/config` endpoint.  I don't know if it is necessary to add.

zafrirron, 2018-04-09:
Thanks @mcspr 

interesting...
Looks like a good starting point, so if I understand you correctly, in order to create a self configuration option, I could enable an AP on any device, that is on any network. and create a static STA on all fresh devices, so in case of fail to connect the device will fallback to try to connect to the espurna "join" network enabled before, get the configuration by accessing /config endpoint. and reconnect to master network...
Does this makes any sense?

zafrirron, 2018-04-10:
@mcspr  thanks for the guidance, will look into it...

mcspr, 2018-04-12:
[Compiled demo app](https://play.google.com/store/apps/details?id=com.cmmakerclub.iot.esptouch)
[IOT Espressif (side menu "add device")](https://play.google.com/store/apps/details?id=com.espressif.iot)
Maybe eWeLink does this too?

But neither of these apps are working on my android :/

xoseperez, 2018-04-12:
Mmm... doesn't sound good. Maybe WPS is a better option, although not all routers have this option...

gn0st1c, 2018-04-13:
for reference:
https://github.com/EspressifApp/EsptouchForAndroid

xoseperez, 2018-09-10:
Both WPS and SmartConfig are available in the last builds. They are not enabled by default, you will have to build your own image with JUSTWIFI_ENABLE_WPS or JUSTWIFI_ENABLE_SMARTCONFIG are build option.

closed 2018-09-10 by xoseperez