xoseperez/espurna #770: Energy measurement on a schedule / frmo MQTT

hyteoo, 2018-04-08:
Coming from issue #671 I think in order to complete this functionality would be to implement an auto reset.

I'm thinking of a real world situation where the kWh would be recorded every month, so a reset at the 1st of any month would make sense.

An elegant way to do this would be to tie into the already existing scheduling engine, however I haven't gone into the code there so I'm not sure how hard that would be.

Another feature would be to reset via a MQTT command.  I'm thinking of having the reset time/day coordinated from a central point for several devices.

bignooz, 2019-01-19:
Would be a Great PLUS. As i use ony mqtt in my home automation system.

Thank you !

bignooz, 2019-03-06:
"@xoseperez xoseperez removed this from the 1.13.5 milestone 10 days ago"

Is this modification cancelled ?

bignooz, 2019-06-08:
Hi!
Has this enhancement been totally cancelled?

CoolID2, 2020-02-11:
Hi,
what's the status of fhis feature request? 
That would be a cool additional function that I could use.

Jessendelft, 2020-12-11:
I'd like to vote for this feature as well. And to add that it should also be possible to do this via the RESTapi (PUT api/energy value=0.0)

mcspr, 2020-12-11:
RE: above, both MQTT (through a .../set topic) and HTTP API should already work with dev builds
Scheduling part is something to add though

sjevtic, 2021-02-22:
Can you elaborate on how to reset the energy counter through the HTTP API?

mcspr, 2021-02-24:
See the `/api/list` output, if you are running latest versions from the dev branch for all available endpoints. There should be `/api/energy` returning current value on GET. Sending `value=0` will reset the energy to 0 (but, only if the sensor supports it)

Jessendelft, 2022-08-19:
Did this feature break in more recent versions?
On version 2021-01-22 I can send an MQTT command as follow:
`ESPURNA/energy/0/set` with a payload 0, and it resets the energy to 0.
On more recent versions, including 2022-06-13 and 2022-05-24, sending this same MQTT command does absolutely nothing

mcspr, 2022-08-19:
> Did this feature break in more recent versions? On version 2021-01-22 I can send an MQTT command as follow: `ESPURNA/energy/0/set` with a payload 0, and it resets the energy to 0. On more recent versions, including 2022-06-13 and 2022-05-24, sending this same MQTT command does absolutely nothing

See #2530 and Releases page in this repo for .bin if you meant nightly release date, MQTT actually causes a crash

closed 2024-08-30 by mcspr