xoseperez/espurna #774: Is there a length limit on the hostname advertised to DHCP?

wsw70, 2018-04-09:
I see that a hostname defined as `chambre1-lampe_bureau` on ESPurna is registered by my DHCP as `chambre1-lampe_burea` (the last `u` is missing).

Wouldn't ESPurna cut the advertised hostname at 20 characters? If so, is there a reason for this? (the DHCP server can certainly handle longer hostnames, I see all kind of `android-97d6cbda92c6fgb56` kind of entries)

mcspr, 2018-04-09:
Yep, thats a bug. Limit should be 32 because hostname is linked to SSID.

20 limit comes from [justwifi](https://bitbucket.org/xoseperez/justwifi/src/cc0016dd77d504a918779dc5bb15ff752a12dc63/src/JustWifi.h?at=master&fileviewer=file-view-default#JustWifi.h-148)
From espurna side https://github.com/xoseperez/espurna/commit/847e1e6390fe874d5fe04fc23a9a2ecdbbd0107e can be reverted and adapted to the new limit.

Skaronator, 2018-04-09:
> 20 limit comes from justwifi

Take a look who wrote this code. It is probably better to increase this to 30 or 40 chars than reverting https://github.com/xoseperez/espurna/commit/847e1e6390fe874d5fe04fc23a9a2ecdbbd0107e

mcspr, 2018-04-10:
@Skaronator What I mean is - it is not possible to go outside 32 with current setup. Access point SSID cannot be longer than that.

Underlying arduino methods already have the limit check for [STA hostname](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L467-L472) and [AP SSID](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L102)  (even better - 31 chars :/) so justwifi can use return values to cover that.

xoseperez, 2018-04-18:
Staged for release. Closing.

closed 2018-04-18 by xoseperez

wsw70, 2018-05-18:
Confirmed that the name is not truncated anymore. Thanks.