xoseperez/espurna #776: Building ESPurna 1.12.5 under Debian Jessie amd64 throws error

sven3, 2018-04-10:
Hi,

building ESPurna 1.12.5 under Debian Jessie amd64 throws the following error:

```
gulp-crass@0.2.2 node_modules/gulp-crass
├── gulp-util@3.0.8 (array-differ@1.0.0, beeper@1.1.1, lodash._reescape@3.0.0, array-uniq@1.0.3, lodash._reevaluate@3.0.0, object-assign@3.0.0, lodash._reinterpolate@3.0.0, replace-ext@0.0.1, has-gulplog@0.1.0, dateformat@2.2.0, vinyl@0.5.3, lodash.template@3.6.2, gulplog@1.0.0, chalk@1.1.3, minimist@1.2.0, multipipe@0.1.2)
├── through2@2.0.3 (xtend@4.0.1, readable-stream@2.3.6)
├── csswring@6.0.3 (onecolor@3.0.5, minimist@1.2.0, postcss@6.0.21)
└── crass@0.12.3 (strong-data-uri@1.0.5, color-convert@1.9.1, svgo@0.7.2)

gulp-htmllint@0.0.14 node_modules/gulp-htmllint
├── through2@2.0.3 (xtend@4.0.1, readable-stream@2.3.6)
├── gulp-util@3.0.8 (array-differ@1.0.0, lodash._reevaluate@3.0.0, beeper@1.1.1, lodash._reinterpolate@3.0.0, array-uniq@1.0.3, lodash._reescape@3.0.0, object-assign@3.0.0, replace-ext@0.0.1, has-gulplog@0.1.0, dateformat@2.2.0, vinyl@0.5.3, minimist@1.2.0, gulplog@1.0.0, lodash.template@3.6.2, multipipe@0.1.2, chalk@1.1.3)
└── htmllint@0.7.2 (promise@8.0.1, bulk-require@1.0.1, htmlparser2@3.9.2, lodash@4.17.5)
+ ./build.sh itead-s20
--------------------------------------------------------------
ESPURNA FIRMWARE BUILDER
Building for version 1.12.6a
--------------------------------------------------------------
Get revision...
--------------------------------------------------------------
Building web interface...
/home/sven/espurna/code/node_modules/gulp-crass/node_modules/csswring/index.js:90
  let c = `#${r1}${r2}${g1}${g2}${b1}${b2}`;
  ^^^

SyntaxError: Block-scoped declarations (let, const, function, class) not yet supported outside strict mode
    at exports.runInThisContext (vm.js:53:16)
    at Module._compile (module.js:373:25)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/home/sven/espurna/code/node_modules/gulp-crass/index.js:6:16)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)

```

Thanks for any hints!
Sven

sven3, 2018-04-13:
Thanks for your response, gn0st1c. I use the nodejs packages from Debian Jessie:
```
$ nodejs --version
v4.8.2
```
Package manager also says it's 4.8.2:
```
$ dpkg -l | grep nodejs
ii  nodejs                            4.8.2~dfsg-1                   amd64        evented I/O for V8 javascript
ii  nodejs-dev                        4.8.2~dfsg-1                   amd64        evented I/O for V8 javascript (development files)
ii  nodejs-legacy                     4.8.2~dfsg-1                   all          evented I/O for V8 javascript (legacy symlink)

```
Thanks!
Sven

xoseperez, 2018-04-13:
Strict mode under 4.8.2 does not work for me either. I have to use 6+. The specific versions I use are  6.11.1 or 8.1.4 (I use nvm to manage node versions).

xoseperez, 2018-04-13:
Strict mode under 4.8.2 does not work for me either, not even with `--use-strict`. I have to use 6+. The specific versions I use are  6.11.1 or 8.1.4 (I use nvm to manage node versions).

sven3, 2018-04-13:
Thanks for the hint xoseperez, so I switched from Debian Jessie to Debian Buster (the latter includes nodejs 8.9.3). But now I face a different error messsage:

```
$ ./build.sh itead-s20
--------------------------------------------------------------
ESPURNA FIRMWARE BUILDER
Building for version 1.12.6a
--------------------------------------------------------------
Get revision...
--------------------------------------------------------------
Building web interface...
module.js:538
    throw err;
    ^

Error: Cannot find module 'through2'
    at Function.Module._resolveFilename (module.js:536:15)
    at Function.Module._load (module.js:466:25)
    at Module.require (module.js:579:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/root/espurna/code/node_modules/gulp-base64-favicon/index.js:5:19)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
```

Although this package is installed:
```
$ dpkg -l | grep through2
ii  node-through2                     2.0.3-1                        all          Make a stream.Transform out of a function - Node.js module
```

Thanks!
Sven

gn0st1c, 2018-04-13:
@sven3: is your npm up to date?

gn0st1c, 2018-04-13:
@xoseperez : thanks for the info. i will recommend nodejs 6+ from now on :)

sven3, 2018-04-13:
@gn0st1c: my npm is version 1.4.21, the most recent in Debian. I tried installing the most current version from https://www.npmjs.com/package/npm , but this ended in another error. :-(

```
$ curl -L https://www.npmjs.com/install.sh | sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  6254  100  6254    0     0  28557      0 --:--:-- --:--:-- --:--:-- 28427
tar=/bin/tar
version:
tar (GNU tar) 1.29
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
install npm@latest
fetching: http://registry.npmjs.org/npm/-/npm-5.8.0.tgz
module.js:538
    throw err;
    ^

Error: Cannot find module '/tmp/npm.19720/package/bin/read-package-json.js'
    at Function.Module._resolveFilename (module.js:536:15)
    at Function.Module._load (module.js:466:25)
    at Function.Module.runMain (module.js:676:10)
    at startup (bootstrap_node.js:187:16)
    at bootstrap_node.js:608:3
added 1 package from 1 contributor in 0.64s
/usr/bin/npx -> /usr/lib/node_modules/npm/bin/npx-cli.js
/usr/bin/npm -> /usr/lib/node_modules/npm/bin/npm-cli.js
+ npm@5.8.0
updated 1 package in 2.574s
It worked
```
The error message when trying to build ESPurna (Cannot find module 'through2') stays the same.

Thanks!
Sven

gn0st1c, 2018-04-16:
sorry, that was all suggestions i have.

xoseperez, 2018-04-17:
It looks like a dependency issue of some of the modules. The only related info I have found is this: https://github.com/robrich/gulp-if/issues/23. ESPurna web builder does not use gulp-if but the solution might be the same...

Anyway this issue is not ESPurna related.

sven3, 2018-04-24:
Thank you very much for your help, @gn0st1c , @xoseperez and @lobradov !

Using lobradov's commands, I managed to compile ESPurna 1.12.5.

Thanks again everybody!
Sven

closed 2018-04-24 by sven3