xoseperez/espurna #783: Request: Schedule ignore daylight savings

BuildTheRobots, 2018-04-11:
Request: 
Have an option for each schedule set to ignore daylight savings.

Reason: 
As I have electric storage heaters in my house (no gas), I get two rate electric. More expensive during the day and cheaper at night to charge the heaters up (called Economy 7 or Economy 10 in the UK).
I'm currently using the sonoff POW modules to switch the heaters on when it's cheap. 
It turns out that even though our daylight savings puts the clocks forward an hour for half the year, the electric company ignores it, so the cheap electric stays tied to UTC-0 rather than being relative. This means in winter it's midnight->7am but in summer it's actually 1am -> 8am.

It would be nice to be able to set a schedule that ignores daylight savings. Without that option I either have to leave a safety buffer of an hour at the start and end (loosing two hours a night) or remember to change the schedule when the clocks change (and who manually resets clocks these days?)

Greatest of thanks :)

xoseperez, 2018-04-11:
You can also disable daylight savings on the device :smile:

BuildTheRobots, 2018-04-11:
I can indeed - the reason I was hoping to do it on the schedule was because I see it as more consistent, especially on multi-switch devices.

The use-case I can see is being able to make sure the heaters switch off all year when the electric gets expensive in the morning (7am UTC-0), but still allow me to have a light that came on when I got back from work at 7pm (which is always 7pm LOCAL).

xoseperez, 2018-04-12:
I've just committed the changes to allow defining a scheduled based on UTC time. It's in the dev branch, let me know if it works for you.

BuildTheRobots, 2018-04-13:
Installed on a POW and seems to be working. Will do some proper testing with multiple timezones tomorrow. Thank you :)

xoseperez, 2018-04-18:
Staged for release. Closing.

closed 2018-04-18 by xoseperez