xoseperez/espurna #784: Request: Energy Usage _from date_

BuildTheRobots, 2018-04-11:
Request: Report date and time when energy consumption was last reset (eg "Energy #0 [40kwh]
HLW8012 @ GPIO(5,14,13), ****since 20180411-1052****")

Having a count of energy consumed is extremely useful, especially now #671 gives us the ability to reset it. However not knowing the time period this applies to makes the data less useful (the difference between that 40kwh being used over the last day, week or month is massive).

Having the unit record when the usage was manually reset would be great; even better if it could notice first boot as well (energy gets reset after firmware update).

Greatest of thanks :)

xoseperez, 2018-04-18:
Staged for release. Closing.

closed 2018-04-18 by xoseperez

BuildTheRobots, 2020-01-18:
After extensive testing, it seems that it's possible for the Energy usage counter on the status page to roll-over (reset to 0) without the date/time of last reset updating.

I'm not entirely sure what the limit it, when it last rolled over, or even how many times it rolled over since the date (this is my hot water tank so might  burn 7kWh/day.

Unrelated, but I'm thrilled by how stable the firmware is - it's last reset is October 2018, it's been stable without any intervention since then :)

mcspr, 2020-01-31:
Just to clarify, timestamp should be from last report (after ~N readings, depending on the setting. webpage only loads it once, it is never dynamically updated). Status page shows the latest reading, so values there the most up-to-date.

What is interesting is that specific limit per HLW library is 2<sup>32</sup> since `unsigned long` is return type of it's `getEnergy()`:
https://github.com/xoseperez/hlw8012/blob/2dace562c339e77a2889446d461e8e56048fe634/src/HLW8012.cpp#L134
Other sensor stuff uses `double` as a container, so bit limit goes to 51 (if I still remember how floating-point).
No overflow checks are in place for that, so it kind-of fills up until it doesn't anymore.