xoseperez/espurna #787: Terminal - Serial port tight coupling

zafrirron, 2018-04-12:
For some reason there is a coupling between terminal and serial port (and in some cases debug and serial port), they should be completely separated:

It should possible not to use serial port at all (not for debug and not for terminal) and use terminal (and/or debug) thru telnet (and/or web).

xoseperez, 2018-04-12:
Setting `DEBUG_SERIAL_SUPPORT` to 0 disables serial port completely (at least from the ESPurna side, other libraries might still use it) but as long as TELNET_SUPPORT and DEBUG_TELNET_SUPPORT are both set to 1 you should be able to use it via telnet.

zafrirron, 2018-04-12:
Thanks

please see settings.ino

```#ifdef DEBUG_PORT
    #define EMBEDIS_PORT    DEBUG_PORT
#else
    #define EMBEDIS_PORT    Serial
#endif
```

Doing some serial stuff without  DEBUG_SERIAL_SUPPORT checking

xoseperez, 2018-04-12:
I see what you mean. It shouldn't be checking SERIAL_BAUDRATE. Actually the only reason to init the Serial object when DEBUG_SERIAL_SUPPORT is 0 is because some sensor or devices uses it (rfbridge for instance), and then the init should be moved to the specific code for that sensor/device.

Same thing for the default EMBEDIS_PORT value, it should be a null pointer and the StreamInjector should handle it.

xoseperez, 2018-04-18:
OK, implementation could be improved (a terminal module?) but the basics are there. No more coupling.

xoseperez, 2018-04-18:
@zafrirron feedback welcome

proddy, 2018-04-22:
this is handy, i was looking for a way around this. On one of my boards I'm using the Rx/Tx of UART0 to read and write values to an external serial bus (connected to my boiler) so have to rely on debug messages to telnet to see whats going on.

closed 2018-05-02 by xoseperez