xoseperez/espurna #788: ESPURNA board - can't keep the calibration values.

Wojcioo, 2018-04-12:
Hello
Again thank You for the job :)
I've uploaded v.08 to espurna board  but I still can't keep the values ​​entered in the calibration permanently.
After using telnet and get pwrRatioC i got this "logs":
$-1
[357475] [WEBSOCKET] Requested action: relay
[357479] [RELAY] #0 scheduled ON in 0 ms
[357482] [RELAY] #0 set to ON
[358485] [RELAY] Saving mask: 1
[361322] [WEBSOC KET] Requested action: relay
[361327] [RELAY] #0 scheduled OFF in 0 ms
[361330] [R ELAY] #0 set to OFF
[362332] [RELAY] Saving mask: 0
[362469] [WEBSOCKET] Requested action: relay
[362473] [RELAY] #0 scheduled ON in 0 ms
[362478] [RELAY] #0 set to ON
[363480] [RELAY] Saving mask: 1

it also lost all stored data after disconnecting from power.
Kind regards

xoseperez, 2018-04-12:
To narrow down the cause of the problem I would try to get more info. Type `info` in the terminal and copy here the result. Also, do a little test like:

```
set testKey hello
get testKey                     [should return "hello"]
[now reboot]
get testKey                     [again, should return "hello"]
```

Wojcioo, 2018-04-14:
info:

[393714]

         [393719] [INIT] ESPURNA 1.12.5
                                       [393720] [INIT] xose.perez@gmail.com
                                                                           [3937                             21] [INIT] http://tinkerman.cat

                               [393722] [INIT] CPU chip ID: 0xF8F4C5
                                                                    [393723] [IN                             IT] CPU frequency: 80 MHz
                         [393724] [INIT] SDK version: 1.5.3(aec24ac9)
                                                                     [393728] [I                             NIT] Core version: 2.3.0
                        [393731] [INIT] Core revision:
                                                       [393734]
                                                                [393736] [INIT]                              Flash chip ID: 0x1440E0
                       [393740] [INIT] Flash speed: 40000000 Hz
                                                               [393742] [INIT] F                             lash mode: UNKNOWN
                  [393745]
                           [393746] [INIT] Flash sector size:     4096 bytes
                                                                            [393                             751] [INIT] Flash size (CHIP):  1048576 bytes
                                             [393756] [INIT] Flash size (SDK):                                1048576 bytes /  256 sectors
                             [393760] [INIT] Firmware size:       484368 bytes /                               119 sectors
             [393766] [INIT] OTA size:           2658304 bytes /  649 sectors
                                                                             [39                             3771] [INIT] EEPROM size:           4096 bytes /    1 sectors
                                                             [393777] [INIT] Emp                             ty space:          16384 bytes /   4 sectors
                                            [393782]
                                                     [393784] [INIT] BOARD: TINK                             ERMAN_ESPURNA_H08
                 [393787] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET                              DOMOTICZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL TH                             INGSPEAK WEB
            [393802] [INIT] SENSORS: HLW8012

                                            [393804] [INIT] Last reset reason: E                             xternal System
              [393808] [INIT] Settings size: 1312 bytes
                                                       [393812] [INIT] Free heap                             : 10000 bytes
             [393816] [INIT] Power: 3196 mV
                                           [393818] [INIT] Power saving delay va                             lue: 10 ms
          [393822]
                   [393823] [WIFI] MODE STA ------------------------------------                             -
 [393829] [WIFI] SSID  Netz
                           [393831] [WIFI] IP    192.168.1.65
                                                             [393834] [WIFI] MAC                                5C:CF:7F:F8:F4:C5
                    [393839] [WIFI] GW    192.168.1.1
                                                     [393840] [WIFI] DNS   192.1                             68.1.1
      [393843] [WIFI] MASK  255.255.255.0
                                         [393846] [WIFI] HOST  ESPURNA_F8F4C5
                                                                             [39                             3850] [WIFI] BSSID C8:0E:14:16:7B:24
                                    [393853] [WIFI] CH    11
                                                            [393855] [WIFI] RSSI                               -66
     [393857] [WIFI] ----------------------------------------------
                                                                   [393863] +OK

Wojcioo, 2018-04-14:
info
[393714]

[393719] [INIT] ESPURNA 1.12.5
[393720] [INIT] xose.perez@gmail.com
[3937 21] [INIT] http://tinkerman.cat

[393722] [INIT] CPU chip ID: 0xF8F4C5
[393723] [IN IT] CPU frequency: 80 MHz
[393724] [INIT] SDK version: 1.5.3(aec24ac9)
[393728] [I NIT] Core version: 2.3.0
[393731] [INIT] Core revision:
[393734]
[393736] [INIT]Flash chip ID: 0x1440E0
[393740] [INIT] Flash speed: 40000000 Hz
[393742] [INIT] F lash mode: UNKNOWN
[393745]
[393746] [INIT] Flash sector size: 4096 bytes
[393 751] [INIT] Flash size (CHIP):1048576 bytes
[393756] [INIT] Flash size (SDK):1048576 bytes /256 sectors
[393760] [INIT] Firmware size: 484368 bytes / 119 sectors
[393766] [INIT] OTA size: 2658304 bytes /649 sectors
[39 3771] [INIT] EEPROM size: 4096 bytes /1 sectors
[393777] [INIT] Emp ty space:16384 bytes / 4 sectors
[393782]
[393784] [INIT] BOARD: TINK ERMAN_ESPURNA_H08
[393787] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNETDOMOTICZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL TH INGSPEAK WEB
[393802] [INIT] SENSORS: HLW8012

[393804] [INIT] Last reset reason: E xternal System
[393808] [INIT] Settings size: 1312 bytes
[393812] [INIT] Free heap : 10000 bytes
[393816] [INIT] Power: 3196 mV
[393818] [INIT] Power saving delay va lue: 10 ms
[393822]
[393823] [WIFI] MODE STA ------------------------------------ -
[393829] [WIFI] SSIDNetz
[393831] [WIFI] IP192.168.1.65
[393834] [WIFI] MAC5C:CF:7F:F8:F4:C5
[393839] [WIFI] GW192.168.1.1
[393840] [WIFI] DNS 192.1 68.1.1
[393843] [WIFI] MASK255.255.255.0
[393846] [WIFI] HOSTESPURNA_F8F4C5
[39 3850] [WIFI] BSSID C8:0E:14:16:7B:24
[393853] [WIFI] CH11
[393855] [WIFI] RSSI -66
[393857] [WIFI] ----------------------------------------------
[393863] +OK

Wojcioo, 2018-04-14:
tests:
set testKey hello
+OK
get testKey
+hello
[764560] [WEBSOCKET] Requested action: reboot
get testKey
$-1

but after reboot is has lost all wifi data, passwords etc......

xoseperez, 2018-04-16:
The flash mode in the debug log before shows "UNKNOWN". Did you select the DOUT flash mode to flash the firmware? If you did the only reason left is a faulty SPI flash chip...

Wojcioo, 2018-04-17:
Here are my flash settings.
[image]

I've got 3 the same ESP12F, and I've flashed all of them, and all they have the same issue? Shall I buy a new one, to flash it again?

Wojcioo, 2018-04-23:
Today I've flashed completely new ESP12F, and there is the same issue :(. What should I do?

xoseperez, 2018-04-23:
Does it also say " [INIT] F lash mode: UNKNOWN" while booting?

Wojcioo, 2018-04-23:
You mean after typeing "info"?
Yes..

xoseperez, 2018-04-23:
There is something wrong in the flashing process... Could you try doing it with esptool directly? Check the [wiki page](https://github.com/xoseperez/espurna/wiki/Binaries).

Wojcioo, 2018-04-24:
[141733] [TELNET] Client #0 connected
                                     info
[146021]

         [146024] [INIT] ESPURNA 1.12.5
                                       [146025] [INIT] xose.perez@gmail.com
                                                                           [1460                  26] [INIT] http://tinkerman.cat

                               [146027] [INIT] CPU chip ID: 0xF66083
                                                                    [146028] [IN                  IT] CPU frequency: 80 MHz
                         [146029] [INIT] SDK version: 1.5.3(aec24ac9)
                                                                     [146033] [I                  NIT] Core version: 2.3.0
                        [146037] [INIT] Core revision:
                                                       [146039]
                                                                [146040] [INIT]                   Flash chip ID: 0x1640E0
                       [146045] [INIT] Flash speed: 40000000 Hz
                                                               [146047] [INIT] F                  lash mode: DOUT
               [146050]
                        [146051] [INIT] Flash sector size:     4096 bytes
                                                                         [146056                  ] [INIT] Flash size (CHIP):  4194304 bytes
                                          [146061] [INIT] Flash size (SDK):   10                  48576 bytes /  256 sectors
                          [146065] [INIT] Firmware size:       484368 bytes /  1                  19 sectors
          [146071] [INIT] OTA size:           2658304 bytes /  649 sectors
                                                                          [14607                  6] [INIT] EEPROM size:           4096 bytes /    1 sectors
                                                          [146082] [INIT] Empty                   space:          16384 bytes /   4 sectors
                                         [146087]
                                                  [146089] [INIT] BOARD: TINKERM                  AN_ESPURNA_H08
              [146092] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DO                  MOTICZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THING                  SPEAK WEB
         [146107] [INIT] SENSORS: HLW8012

                                         [146109] [INIT] Last reset reason: Exce                  ption
     [146113] [INIT] Settings size: 1239 bytes
                                              [146117] [INIT] Free heap: 9912 by                  tes
   [146120] [INIT] Power: 2960 mV
                                 [146122] [INIT] Power saving delay value: 10 ms
                                                                               [                  146126]
        [146127] [WIFI] MODE AP --------------------------------------
                                                                      [146133] [                  WIFI] SSID  ESPURNA_F66083
                          [146136] [WIFI] PASS  44prezydentUSA
                                                              [146139] [WIFI] IP                      192.168.4.1
               [146144] [WIFI] MAC   5E:CF:7F:F6:60:83
                                                      [146146] [WIFI] ----------                  ------------------------------------
                                    [146151] +OK
                                                ^C[197002] [WEBSOCKET] Parsing configuration data
                                                                                                 [200430] [WEBSOCKET] Requested action: relay
                                           [200433] [RELAY] #0 scheduled ON in 0 ms
                                                                                   [200438] [RELAY] #0 set to ON
              [201441] [RELAY] Saving mask: 1
                                             [201788] [WEBSOCKET] Requested action: relay
                                                                                         [201791] [RELAY] #0 scheduled OFF in 0 ms
                                [201795] [RELAY] #0 set to OFF
                                                              [202797] [RELAY] Saving mask: 0
                                                                                             [242328] [WEBSERVER] Request: GET /config

Wojcioo, 2018-04-24:
Today I've uploaded firmware using esptool, and there are the same issues :(, but [217395] [INIT] Flash mode: DOUT

[214682] [TELNET] Client #0 connected
info
[217372]

[217372] [INIT] ESPURNA 1.12.5
[217373] [INIT] xose.perez@gmail.com
[217373] [INIT] http://tinkerman.cat

[217373] [INIT] CPU chip ID: 0xF66083
[217374] [INIT] CPU frequency: 80 MHz
[217377] [INIT] SDK version: 1.5.3(aec24ac9)
[217382] [INIT] Core version: 2.3.0
[217384] [INIT] Core revision:
[217387]
[217388] [INIT] Flash chip ID: 0x1640E0
[217392] [INIT] Flash speed: 40000000 Hz
[217395] [INIT] Flash mode: DOUT
[217398]
[217399] [INIT] Flash sector size:4096 bytes
[217404] [INIT] Flash size (CHIP):  4194304 bytes
[217408] [INIT] Flash size (SDK):1048576 bytes /  256 sectors
[217413] [INIT] Firmware size:  484368 bytes /  119 sectors
[217419] [INIT] OTA size: 2658304 bytes /  649 sectors
[217425] [INIT] EEPROM size: 4096 bytes / 1 sectors
[217430] [INIT] Empty space:16384 bytes /4 sectors
[217436]
[217437] [INIT] BOARD: TINKERMAN_ESPURNA_H08
[217440] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DOMOTICZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB
[217454] [INIT] SENSORS: HLW8012

[217457] [INIT] Last reset reason: External System
[217461] [INIT] Settings size: 294 bytes
[217465] [INIT] Free heap: 13624 bytes
[217469] [INIT] Power: 2970 mV
[217471] [INIT] Power saving delay value: 10 ms
[217475]
[217476] [WIFI] MODE AP --------------------------------------
[217481] [WIFI] SSID  ESPURNA_F66083
[217485] [WIFI] PASS  ***********
[217488] [WIFI] IP 192.168.4.1
[217492] [WIFI] MAC5E:CF:7F:F6:60:83
[217494] [WIFI] ----------------------------------------------
[217500] +OK

Wojcioo, 2018-04-27:
Xose, have You got any ideas :) ?

rwagnervm, 2018-05-01:
I'm having the same problem

xoseperez, 2018-05-02:
I can't see anything wrong with the latest log... You should be using flash size 4MB instead. It should also work with 1MB but maybe it's worth trying it...

Wojcioo, 2018-05-04:
I've uploaded with 4MB, and now I can save SSID and password for my wifi.
But when I entered these values
[image]
and I've clicked "save" they disappeared.
In the status section there is only voltage.
[image]

xoseperez, 2018-05-08:
I'm afraid 14W might be just there, in pair with the sensitivity/noise level of the power monitor chip...  Can you test it with a ~60W minimum load?