xoseperez/espurna #791: Feature Request: Configurable transition time

CrispinP, 2018-04-13:
Could we have a transition time for when a change is sent it transitions over n number of ms.

Use case:
I have LED strips up-lighting the ceiling. At night they dim down to almost nothing.
When motion is detected when everyone has gone to bed I tell them to bright up (kids have come downstairs to the kitchen or toilet etc)
With the current implementation they change to quickly. I would like to be able to change the duration they take to brighten up.

CrispinP, 2018-04-20:
Sorry, did not see your reply...
I'll take a look next week and see. Should all be good then :)

FWDInnovations, 2018-07-18:
To add to this request, I would love to see the option to define a custom transition time along with the RGB&W values to be able to control those per transition. For example, when turning the lights on or off, I might want them to fade in and out in 5 seconds, but when changing between 'scenes' while on, I may what the transition time to be 2s. So when sending the new values, you would send something like { R, G, B, W, T} Where T = transition time in ms.

closed 2018-09-23 by stale[bot]