xoseperez/espurna #796: How to use mqttGroup ? (2 and 3-way swiches)

LubergAlexander, 2018-04-14:
I would like to set up several 2-way and maybe 3-way switches. I found a mention of mqttGroup parameter, but not sure where and how to configure it. It doesn't look exposed on the UI.

Is any special wiring required in this case or grouping switches in an mqtt group will do it?

My assumption is that only 1 switch out of 3 needs to trigger a relay, others should only send mqtt messages.

Skaronator, 2018-04-14:
What switches exactly? MQTT Group only works for light AFAIK. Also it looks like it only work with RGB.

https://github.com/xoseperez/espurna/blob/6f190f6d26418ecd359a504e10a572fe2c7fd08e/code/espurna/light.ino#L578-L585

LubergAlexander, 2018-04-14:
@Skaronator I also see it used in code/espurna/relay.ino.

I have regular light switches. I would like to group relays, in order to have 2-way switch setup working.

mcspr, 2018-04-14:
@LubergAlexander But it is in WebUI. Look at *Switches* panel, every switch has *MQTT Group* text field. Make it the same on every device you want to sync.

There's also grouping mechanism for on-device relays, same panel.

LubergAlexander, 2018-04-14:
@mcspr Thanks alot! somehow I've managed to not check the most obvious place.

closed 2018-04-14 by LubergAlexander