xoseperez/espurna #798: Support for Supla devices

jsponz, 2018-04-16:
Hi, 

is there any intention to support supla/zamel devices? 

https://www.supla.org/en/download

I mean devices such as ROW-01, ROW-02, etc.

Thanks!

xoseperez, 2018-04-16:
It's not obvious what microcontroller they use. It looks like they provide ESP8266 code so you can create your own device and connect to their home automation system, but they are not saying what MCU they use in their own products and even less providing a schematic.

I guess the only option it to get one and tear it apart. Do you happen to have a spare one you can open?

jsponz, 2018-04-17:
They have a ESP-WROOM-02 module.

xoseperez, 2018-04-17:
That's good. We still need to know what GPIOs go where. I have asked them for this info.