xoseperez/espurna #802: No detection of external buttons

jsponz, 2018-04-20:
I have a prototype, which uses two external buttons to activate two relays independently. 
External buttons could be switches or push buttons. They are connected to two inputs of the ESP. 
It is connected to a MQTT broker. I can activate the relays via MQTT and I get the relay status change via MQTT as well. 
Problem: I can't activate the relays via external buttons. My configuration is:

    // Info
    #define MANUFACTURER        "Acme"
    #define DEVICE              "123"

    // Buttons
    #define BUTTON1_PIN         14 
    #define BUTTON1_MODE        BUTTON_SWITCH 
    #define BUTTON1_RELAY       1 
    #define BUTTON2_PIN         12
    #define BUTTON2_MODE        BUTTON_SWITCH 
    #define BUTTON2_RELAY       2 

    // Relays
    #define RELAY1_PIN          13 
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
    #define RELAY2_PIN          15 
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1

On the pins of the ESP, I get the signal when I click on the external button, but nothing happens: no actions on the relays or message to the MQTT server. 

What am I doing wrong? 

Thanks for your help!

xoseperez, 2018-04-23:
The configuration looks fine. I suppose your switches set the pins to either HIGH or LOW, no floating, right? I'll have to test it myself.

xoseperez, 2018-06-04:
@jsponz I cannot reproduce this issue. A debug log of the device while clicking the buttons will help...