xoseperez/espurna #803: Scheduled restart?

oskarirauta, 2018-04-20:
I **STILL** have connectivity issues - every week, multiple devices loose connection to access points and they never re-connect.

One of these devices (POW) is in proximity of 3 meters from access point. And it seems - that more you use the device (switch on and off) - more often this happens. This POW I mentioned is used to control power outlet which is used by my spouse to pre-heat her car, and because it's still cold, this is every day. I have scheduled it, but this schedule is elsewhere, it's done with home assistant, because I have created a script, that turns pre-heating off always after it has been on for 3 hours, even when started manually with the button.

When device has lost this connection, I can connect to AP created by device itself, and press either re-connect, or reboot and then it's possibly usable again. Most of the times, reconnecting only renders device to not connect to Wifi, and to not make AP either (in this case, I need pull the fuse to manually restart it).

Anyway, as this seems to be very difficult to debug or replicate according to previous conversations ( at the old site ) - I have come to think about a work around solution for this.

Now when there's scheduling-
Could we get some more options? Instead of controlling relay, I'd like to control the device itself, and here are my ideas about missing options that possibly could _solve_ my issue...

- reboot (forced)
- reboot if relay(s) is/are off (there might be sensitive device which relay is controlling)
- re-connect

And we have discussed about this issue before- there's some sort of automatic functionality that re-connects to chosen AP after loss of connection/error/etc... I have been trying to find that part of code to see if I could do something about it (probably not, you seem pretty capable), but could you give me some pointers to the code, where are located following parts:
 - automatic wifi reconnection
 - test for reboot necessity (stability testing)

Thanks :) Still, even though I have these issues; great work, this is most certainly the best firmware available :)

I have included a screenshot of available wifi AP's after 5 days ( I drive a truck, so I am not at home at the week, and idea is that I have control over my devices through VPN from distance )
As you see, most devices are once again in the state of self-AP.

[image]

oskarirauta, 2018-04-20:
For some reason, one of the devices decided to use it's hostname as AP name, instead of ESP_*
(bedroom-switch)

Is this related to the issue? Devices are all Sonoff's:
 - sonoff basic
 - sonoff pow
 - sonoff touch
 - sonoff T1 EU 2 gang
 - sonoff s20

And after every week mileage varies, most times, atleast 2 devices have lost their connection.

xoseperez, 2018-04-23:
It's a good idea but it's also very bad news you are having all these troubles with all (?) of your devices. There must be something going on. One question: if I understood you correctly, you can still connect to the device's own AP, right? Have you noticed if it was running in safe mode? What was the uptime of the device?

icevoodoo, 2018-04-23:
If the problem regarding connectivity persist, a schedule for reboot is a must.
We need this option in schedule section. From my site of view I vote for this.

+1

law1964, 2018-05-15:
@oskarirauta Do you still have this connectivity issue? I had a Belkin wemo switch before that would not connect to my wifi network. I changed the wifi mode from 802.1n only to 802.1g/n mixed and it started connecting. I don't know what your wifi router settings are, but if it is restricted to "n" then, perhaps try changing the mode to "g/n" or even "b/g/n" to determine whether it is a compatibility issue. Not all wifi routers are built the same.

oskarirauta, 2018-05-15:
I got issue worked out by compiling against newer Core version.

xoseperez, 2018-05-15:
@oskarirauta Can you confirm it was due to Core 2.4.0? 2.4.1 works fine for you? What lwIP version you compiled against?

mcspr, 2024-08-22:
Closing via #2417, impl. changed to use terminal commands

closed 2024-08-22 by mcspr