xoseperez/espurna #805: Unable to get Minimal Firmware Page after SonOTA

seetendra, 2018-04-22:
Hello Everyone,

I was just trying SonOTA using the instruction given in the Espurna Wiki.
Was able to get it working till where i see ESPURNA_XXXXXX and iam able to connect to it as wall using fibonacci as the password
Once iam connected iam not able to go any further.
I dont't get the webpage where i need to enter the SSDI and Password.
What am i doing wrong or it a Bug ?

lobradov, 2018-04-22:
Are you opening http://192.168.4.1 after you connect to wifi? 

If so, are you, by any chance, using iPhone and Safari? There's a known bug in WS authentication, so you might need to try with Chrome or another non-iOS device.

seetendra, 2018-04-22:
Hi

No iam not using a iphone i tried using a windows with chrome and also a Mac with Chrome
 No luck with both.

seetendra, 2018-04-22:
I get page not found

xoseperez, 2018-04-23:
I would first double check you are indeed connected to the device AP, check you have 192.168.4.X IP in your phone. Also, sometimes the IP/net that the ESP8266 creates is not in the 192.168.4.0/24 range but in the 192.168.244.0/24, in this case the IP of the device would be 192.168.244.1. 

That reminds me that it would be nice to have captive portal enabled when in AP mode...

lobradov, 2018-04-23:
+1 

Was thinking about redirection while in AP mode...

seetendra, 2018-04-23:
Iam at work right now will try it once i reach back home.
@xoseperez Thx a lot for your input.

seetendra, 2018-04-25:
@xoseperez @lobradov 
Guys i tried the steps and iam not able to connect with 192.168.244.1 and 192.168.4.1 as well
Iam using Chrome on a Mac and Yes i have connected to the Device with IP 192.168.4.4

[image]

Is there a Way i can Telnet to the device and then get it connected to My Router.
If yes what is the username and password for login i have already tried admin and fibonacci and it does not connect.

Also with 192.168.244.1 i don't get an error it i just not loading the page.

mcspr, 2018-04-25:
I noticed that you did use SonOTA method. Did you flash the second firmware? Core does not have WebUI enabled.

`Router:` is the esp device IP, just `telnet 192.168.4.1` without any user/password things

mcspr, 2018-04-25:
Mmm you got the firmware you want, right? https://github.com/xoseperez/espurna/wiki/OTA describes manual OTA updates using platformio and source tree of espurna.

Or, you can try *releases* page here and use script provided by Arduino/esp8266:
https://raw.githubusercontent.com/esp8266/Arduino/91833667343b17d4494464e80ca900eae884bff8/tools/espota.py
get this script saved somewhere with relevant firmware binary present in the same directory, then run Terminal.
`python espota.py -i 192.168.4.1 -f <firmware-filename> -a fibonacci -r`

seetendra, 2018-04-26:
One good thing is i was able to Telnet Today to the device

Trying 192.168.4.1...
Connected to 192.168.4.1.
Escape character is '^]'.
[863360] [TELNET] Client #0 connected
[900315] [MAIN] Uptime: 900 seconds
[900316] [MAIN] Free heap: 35224 bytes
[900317] [MAIN] Power: 3156 mV
^]
telnet> 

So is there anything i can do now to flash the Main Bin file from Telnet

seetendra, 2018-04-26:
@mcspr No i have downloaded the espota.py file and the running it with the command.
Attached is the file that iam using 
[espota.py.zip](https://github.com/xoseperez/espurna/files/1952434/espota.py.zip)

seetendra, 2018-04-26:
@xoseperez Any help here?

mcspr, 2018-04-26:
@seetendra You are correct that it has .py extension, but you still got the wrong file and need to re-download it.
Try in Terminal - `open -e espota.py`

seetendra, 2018-04-26:
Let me try that right now

seetendra, 2018-04-26:
That Work like a Charm. Your the Best buddy now i need to do a write up on this on the Wiki do that we can share the learning.

closed 2018-04-27 by seetendra