xoseperez/espurna #819: Feature request: scheduler by relative time

obstler42, 2018-04-26:
I would like to schedule commands not by the absolute real time, but by relative time since startup/bootup.

For example:

5 minutes after startup turn on relay
10 minutes after startup turn off relay

I tried "hacking" the scheduler and NTP configuration by inputting a non existing IP address for the NTP server (there is no option to turn NTP off?), and setting a timer for 0 hours 5 minutes etc., but this doesn't seem to work as the board apparently never gets a time without NTP (maybe it should simply start at 00:00 when there is no valid NTP..)

obstler42, 2018-04-27:
In the meantime here is a little hack on how to achieve a relative time schedule after bootup when you don't need the real time on the espurna device:

in the file code/espurna/ntp.ino change
`bool ntpSynced() {
    return (year() > 2017);
}`
to
`bool ntpSynced() {
    return true;
}`

build the image and flash it to the device.

in the config for NTP make sure the NTP Server field is empty, set timezone to GMT+0 and DST to off.

Now whenever the Espurna device boots it will set the time to 1970-01-01 00:00 and you can simply use the scheduler to turn on the switch at 00:05 (five minutes after bootup) and turn off at 00:10 (ten minutes after boot) etc.

Still looking forward to an official implementation of that feature that also works with real NTP time set, but for now I can use the relative schedule hack and still have all the nice espurna features.

cheers!

sfilinto, 2018-11-30:
I have a similar ( but simpler & single ) condition of turning a switch OFF after 10 minutes of boot time. I am able to achieve this by setting the pule time to 600 & Pulse mode to Normally OFF.

closed 2024-08-22 by mcspr