xoseperez/espurna #826: Feature request: Auto Disable Wifi

tty74it, 2018-04-29:
It would be useful to turn off the wifi in the t1 touch sonoffs installed in the bedroom near the head of the bed, so that during the night you avoid having additional wifi waves, of course maintaining the functionality of the switch. As soon as the button is pressed, the wifi is re-locked and then automatically disabled.
 like this project:
https://github.com/omersiar/esp-rfid
https://bitadvise.com/esp-rfid/

Thanks and congratulations for the espurna project

gretel, 2018-04-29:
@tty74it you should think about talking to your neighbours about not using their microwave ovens when you get to sleep, and maybe have them get some sonoff relays, so you'd switch them. safety first!

xoseperez, 2018-05-02:
Anyway @tty74it suggestion is legit. The latest JustWifi version has WIFI_OFF mode so it should be possible to simply map it to a certain number of button clicks...