xoseperez/espurna #829: Sonoff T1

seetendra, 2018-04-29:
Hello Guys
Iam trying getting the sonoff T1 3 Gang Uk version to work with Espurna
It had Tasmota in it which i flashed using SonOTA 
But it was my bad i flashed the wrong firmware instead of Sonoff T1 i flashed the Sonoff Touch
and it stopped working. (this was when i was moving from Tasmota to Espurna)
I have now flashed again esptool.exe 
but now it is not showing me the SoftAPmode any more it is just a blue light blinking and nothing else it happening.
When i press and hold the button 1 continuous  i get 1 beep and then 2 beep and the the button goes off.
What am i doing wrong?
BTW the buttons work fine now after the flash with serial else it was just the First button.

xoseperez, 2018-05-02:
Do you still have access via serial? If so try getting the debug log to see what's going on. You can use terminal commands to configure the wifi credentials from the console. Check this: https://github.com/xoseperez/espurna/wiki/Terminal

seetendra, 2018-05-03:
Sure let me try and i will get back to you with the logs

seetendra, 2018-05-20:
@xoseperez  I just tried to flash the Ti again and this time iam able to get all the buttons to work but for some reason iam not able to get device to come in SoftAP mode to configure it
Looks very strange.

xoseperez, 2018-05-21:
Yes, the logs can help find out what going wrong...

seetendra, 2018-05-31:
Any Guide on how to do a log capture using serial

xoseperez, 2018-06-04:
Just connect the wire like you did when you first flashed it and then open a terminal session to it. You can use the Arduino IDE "serial monitor" or other equivalent tools like minicom, picocom, screen,... for linux/mac. If using a different OS search for serial monitor tools. Whatever tool you use, ensure you set it to 115200 baud.

seetendra, 2018-06-04:
Ok i will give it a try right now
I use Atom on a mac most of the times.

seetendra, 2018-06-26:
@xoseperez  Hey sorry for the delay in getting back to you with the details
Here you go

This the is error iam getting 

epc1=0x4022db40, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0):

xoseperez, 2018-06-26:
Well I will need a little more info. If you can capture the log just before the exception and the exception stack it could be a start point...

seetendra, 2018-06-26:
I don't see any thing except this and this keeps repeating all over again.
I did try the reset command 
I will check again once i reach back home and see if we can get some more info

seetendra, 2018-06-29:
Please close the issue
Iam going make some update in the WIKI

closed 2018-06-30 by seetendra