xoseperez/espurna #830: H801 - 5 channel workaround doesn't work anymore

flashy02, 2018-04-30:
Hi,

i've tried the new versions 1.12.5 master /and 1.12.6a dev today. The problem is, that the workaround to use 5 channels seperately with buttons doesn't work annymore. With 1.12.4 everything is fine.

I've defined the following in hardware.h as everytime:

define DUMMY_RELAY_COUNT   5
    // Buttons
    #define BUTTON1_PIN         3
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    #define BUTTON2_PIN         0
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON2_RELAY       2

    #define BUTTON3_PIN         2
    #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON3_RELAY       3

    #define BUTTON4_PIN         5
    #define BUTTON4_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON4_RELAY       4

    // Relays
    #define RELAY1_PIN          15
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    #define RELAY2_PIN          13
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL

    #define RELAY3_PIN          12
    #define RELAY3_TYPE         RELAY_TYPE_NORMAL

    #define RELAY4_PIN          14
    #define RELAY4_TYPE         RELAY_TYPE_NORMAL

Best regards
Michael

xoseperez, 2018-05-02:
Using DUMMY_RELAY_COUNT with RELAY#_PIN is not compatible, only the 5 dummy relays should be created.

flashy02, 2018-05-02:
Should I only define the buttons? At the moment, the posted cutout of my hardware.h works fine. I would try the following with 1.12.5:

#define DUMMY_RELAY_COUNT   5
// Buttons
#define BUTTON1_PIN         3
#define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
#define BUTTON1_RELAY       1

#define BUTTON2_PIN         0
#define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
#define BUTTON2_RELAY       2

#define BUTTON3_PIN         2
#define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
#define BUTTON3_RELAY       3

#define BUTTON4_PIN         5
#define BUTTON4_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
#define BUTTON4_RELAY       4

flashy02, 2018-05-02:
Hi, i've tested this version with the new 1.12.6, but it still doesn't work. If I define more then 1 DUMMY and the named buttons, there is still the Problem that no channel reacts on pushing a button (HW + SW) and all channels permanently light up.

ColinShorts, 2018-05-07:
@flashy02, any chance you can expand on your use case for 5 relays on the h801, and why you have buttons defined? I've got a spare unit I can test with, but useful to know how to hook it up for testing.

If you are using the h801 as a 5 channel relay controller with no dimming, it would make more sense to look to the way that the "SONOFF_4CH" is setup, but unless you have added physical buttons to the h801 (seems unlikely), you would only define the relays.

HTH

Colin

flashy02, 2018-05-08:
Hi Colin, at the moment I use one H801 to controle 4 different light channels (stripes, spots) in my house installation. Additionaly I can controle these channels with my "old" push-button installation, so I have soldered some cables to the defined pins. I ve decided to the H801 explicitly because of the dimming and fading function. 
At the moment it works fine with 1.12.3 and 1.12.4, but since 1.12.5 something has happend, because if I define more then 1 dummyrelay, all channels permanently light up after flashing without any reaction on pushing the HW or SW buttons.

Best regards
Michael

ColinShorts, 2018-05-08:
Hi Michael, I'm not quite ready to break out the fine-tipped soldering iron, so I was limited to testing using GPIO0 and after disabling the WiFi LED, it seems to work ok for me when compiled/flashed using platformio.

https://github.com/ColinShorts/espurna/tree/huacanxing-h801-flashy02

[firmware.bin.gz](https://github.com/xoseperez/espurna/files/1985792/firmware.bin.gz)

Feel free to compile yourself or try the attached binary.

-Colin

flashy02, 2018-05-10:
Hi Colin, sorry but it doesn't work. I've flashed your bin and the same fault. I can push the HW-button or the SW-Button but only the state of the button on the webpage changes. The light channels permanently light up with no reaction.

ColinShorts, 2018-05-10:
Hi Michael, I can mostly confirm what you are seeing after swapping in the unit - at least with my version as above. It was already past my bedtime by the time I tested the unit without external lights. Just now I can switch all the lights on and off using the first switch (after adjusting the dimmer to be above 0), but none of the extra relays do anything.

I've tried a few different angles today, but I think the only sensible way I could probably figure it out is to checkout the code to specific tags, compile, flash, and test. There have been a lot of changes since 1.12.4, so that would take a long time. If you were to ignore the buttons for now, you should be able to set the number of dummy relays to 5 (or 6 if you want a master off switch - see the comments in relay.ino).

Does @xoseperez have any comment on the best way to test/proceed with this?

ColinShorts, 2018-05-10:
@flashy02 @xoseperez  looks like I've managed to track down where it went wrong at least...

04b23cd1466f124872bc6d5a9e403ce75b3357d8 (6/4/18) ok
2bc1e6f97ab52dcf91d7c7722374d5ac7e16b455 (7/4/18) bad

I checked another 12 days before the 6th, but as they all worked there's probably not much point in wasting pixels here.

time for bed now methinks, hope this helps track down the solution!

flashy02, 2018-05-12:
@Coli: Thank you for searching the right version. I hope it would help  to implement the workaround again.

xoseperez, 2018-05-13:
Thank you @ColinShorts. I found the regression and fixed it in dev branch.

flashy02, 2018-05-13:
Hi, i've tested it. It works like before. Thank you @xoseperez .

xoseperez, 2018-05-13:
Nice. Closing, will be released with 1.12.7

closed 2018-05-13 by xoseperez