xoseperez/espurna #831: Feature request: api for open relay for n miliseconds

dddsssasa, 2018-04-30:
I would like use sonoff basic for electric door lock. If I use standard api for relay {root topic}/relay/0/set and call it by complicated central unit (raspberry j2ee) which connected throw not 100% reliable wifi, the system could fall and the lock would remain open.

Best solution should be api(mqtt, web) such as 
{root topic}/relay/0/pulse  => { 5000}

wildwiz, 2018-04-30:
maybd you want to use the pulse feature ? on your espurna, go to switches, choose pulse mode normally off and give it a pulse time of, say, 0.25 s ?

dddsssasa, 2018-04-30:
eventuelly yes, but api would be more comfortable. Some clients also wont short and long pulse for door lock.

xoseperez, 2018-06-12:
Thanks to @mcspr  for #902. I've added the HTTP API entry point to his work. Now you can pulse any relay for an arbitrary time. To toggle the relay and toggle it back after 10 seconds just send this request (notice times are in milliseconds):

```
curl "http://192.168.1.123/api/pulse/0?apikey=AABBCCDDEEFF0011&value=10000"
```

or using MQTT:

```
mosquitto_pub -t ESPURNA_001122/pulse/0/set -m 10000
```

xoseperez, 2018-06-12:
Staging for release

closed 2018-06-12 by xoseperez