xoseperez/espurna #837: PZEM004T compiling code problem

MariusLaurinaitis, 2018-05-02:
Hello, 
I have problem with PZEM004T, Platformio dont compile the code, I get an errors: 
espurna/config/../sensors/PZEM004TSensor.h:131:23: error: expected identifier before numeric constant
espurna/config/../sensors/PZEM004TSensor.h:131:23: error: expected ',' or '...' before numeric constant
espurna/config/../sensors/PZEM004TSensor.h:130:28: error: invalid initialization of non-const reference of type 'Stream&' from an rvalue of type 'int'
espurna/config/../sensors/PZEM004TSensor.h:23:41: error: class 'PZEM004TSensor' does not have any field named '_data'
espurna/config/../sensors/PZEM004TSensor.h:75:25: error: no match for 'operator==' (operand types are 'Stream' and 'int')
espurna/config/../sensors/PZEM004TSensor.h:76:23: error: cannot convert 'PZEM004T' to 'PZEM004T*' in assignment
espurna/config/../sensors/PZEM004TSensor.h:78:41: error: no matching function for call to 'PZEM004T::PZEM004T(Stream&)'
espurna/config/../sensors/PZEM004TSensor.h:80:34: error: no matching function for call to 'PZEM004T::setAddress(<unresolve
d overloaded function type>)'
espurna/config/../sensors/PZEM004TSensor.h:101:23: error: '((PZEM004TSensor*)this)->PZEM004TSensor::_ip' does not have cla
ss type
espurna/config/../sensors/PZEM004TSensor.h:115:54: error: no matching function for call to 'PZEM004T::current(<unresolved 
overloaded function type>)'
espurna/config/../sensors/PZEM004TSensor.h:116:54: error: no matching function for call to 'PZEM004T::voltage(<unresolved 
overloaded function type>)'
espurna/config/../sensors/PZEM004TSensor.h:117:52: error: no matching function for call to 'PZEM004T::power(<unresolved ov
erloaded function type>)'
espurna/config/../sensors/PZEM004TSensor.h:118:53: error: no matching function for call to 'PZEM004T::energy(<unresolved o
verloaded function type>)'

MariusLaurinaitis, 2018-05-09:
Thanks, 
I downloaded DEV, BIN compiled, but WEB GUI does not turn on. I'll wait, maybe I hurried up

Marius

xoseperez, 2018-05-09:
Yeah, just fixed the compiling problems, I have not test it yet.

MariusLaurinaitis, 2018-05-22:
Hello, any news?

Thanks
Marius

xoseperez, 2018-05-31:
I've been testing the PZEM with ESPurna and there an issue when using software serial that I have not been able to solve. Using hardware serial works just fine. I have changed the default method in the `sensors.h` file for the moment. Notice that using hardware serial for PZEM is not compatible with serial debug, so to build an image with this settings the hardware definition must state:

```
#define PZEM004T_SUPPORT                0
#define DEBUG_SERIAL_SUPPORT            0
```

xoseperez, 2018-05-31:
Actually, using an oscilloscope shows that the PZEM is not answering back when using software serial... not sure the reason, but it does when using hardware serial..

xoseperez, 2018-06-12:
@MariusLaurinaitis Can you give it a try using hardware serial?

MariusLaurinaitis, 2018-06-13:
Hello, I tried it yesterday. But I do not understand. I have downloaded the DEV version. What do I need to change in specific files? At PZEM I connect the Wemos mini.

MariusLaurinaitis, 2018-06-13:
arduino.h
#define WEMOS_D1_MINI_RELAYSHIELD
#define DEBUG_SERIAL_SUPPORT   0
#define PZEM004T_SUPPORT       1

sensors.h
//------------------------------------------------------------------------------
// PZEM004T based power monitor
// Enable support by passing PZEM004T_SUPPORT=1 build flag
//------------------------------------------------------------------------------

#ifndef PZEM004T_SUPPORT
#define PZEM004T_SUPPORT                0
#endif

#ifndef PZEM004T_USE_SOFT
#define PZEM004T_USE_SOFT               0       // Use software serial
#endif

#ifndef PZEM004T_RX_PIN
#define PZEM004T_RX_PIN                 13      // Software serial RX GPIO (if PZEM004T_USE_SOFT == 1)
#endif

#ifndef PZEM004T_TX_PIN
#define PZEM004T_TX_PIN                 15      // Software serial TX GPIO (if PZEM004T_USE_SOFT == 1)
#endif

#ifndef PZEM004T_HW_PORT
#define PZEM004T_HW_PORT                Serial1 // Hardware serial port (if PZEM004T_USE_SOFT == 0)
#endif

MariusLaurinaitis, 2018-06-13:
I understand that I can specify the Wemos PIN:
hardware.h
 
 // PZEM004T
    #define PZEM004T_SUPPORT    0   // not working?
    #define PZEM004T_RX_PIN     3
    #define PZEM004T_TX_PIN     1

xoseperez, 2018-06-13:
No, it should be like this:

```
#define PZEM004T_SUPPORT                1
#define PZEM004T_USE_SOFT               0
#define PZEM004T_HW_PORT                Serial
#define DEBUG_SERIAL_SUPPORT            0
```

MariusLaurinaitis, 2018-06-13:
OK,
#define PZEM004T_HW_PORT                Serial 

#define PZEM004T_HW_PORT                Serial1 ? (delete 1?)

And Wemos PIN RX and TX? Do I need to use the standard Wemos RX TX?

MariusLaurinaitis, 2018-06-13:
[image]

xoseperez, 2018-06-13:
The main hardware port in the ESP8266 uses GPIOs 1 and 3. These are mapped to RX and TX in the Wemos. SO yes, you have to wire the PZEM004T to RX/TX and set `PZEM004T_HW_PORT` to `Serial`.

MariusLaurinaitis, 2018-06-14:
[image]

MariusLaurinaitis, 2018-06-14:
OK!
as I understood, the SWITTCH#0  must be ON

xoseperez, 2018-06-14:
Nice, I'm labeling it as ready to deploy

closed 2018-06-14 by xoseperez

1043717432, 2018-08-06:
Hey, what's the final configuration?

rradar, 2019-02-17:
Is there a chance to get a pre build image for the pzem004t and a wemos d1 mini or the node mcu?

Also wiring would be interesting... :smiley:

jen101121, 2024-03-16:
hey, goodafternoon, I need help, Can you help me? 
my problem is, arduino ide software, said "no matching for call to pzem004tv30", how can i solve this?

mcspr, 2024-03-16:
@jen101121 update ESP8266 Core in boards manager, it must be 3.1.2
don't see any problems when using arduino-ide 2.x or arduino-cli

If you still have any issues, **create** a new one instead of continuing here

jen101121, 2024-03-16:
Ohh, I see, I try later, Thank you. I'll update you if it is working.