xoseperez/espurna #838: BNSZ01 Switch #0  on/off after upgrade to 1.12.6 stop working

tty74it, 2018-05-02:
Channel #0 is working with no problem from 0 to 255

Thank's

xoseperez, 2018-05-08:
I guess you meant it's NOT working. Based on your comment I assume you did test it with previous versions and it worked. Can you log via telnet to the device and check the debug log?

tty74it, 2018-05-09:
Yes, i've also tried the releases binary on gihub, but is the same.
This is the log
[640452] [WEBSOCKET] #2 connected, ip: 192.168.88.223, url: /ws
[709628] [WEBSOCKET] Requested action: relay
[709632] [RELAY] #0 scheduled OFF in 0 ms
[709637] [RELAY] #0 set to OFF
[709639] [MQTT] Sending tavlucecentro2/channel/0 => 255 (PID 1)
[709643] [MQTT] Sending tavlucecentro2/relay/0 => 0 (PID 1)
[710644] [RELAY] Saving mask: 0
[715574] [WEBSOCKET] Requested action: relay
[715577] [RELAY] #0 scheduled ON in 0 ms
[715579] [RELAY] #0 set to ON
[715582] [MQTT] Sending tavlucecentro2/channel/0 => 255 (PID 1)
[715586] [MQTT] Sending tavlucecentro2/relay/0 => 1 (PID 1)
[716587] [RELAY] Saving mask: 1

I've click on web interface on->off and off->on Switch #0
but the light is always on

Thank's
Tiziano

xoseperez, 2018-05-13:
Probably the same issue as #830 . A fix has been pushed to dev branch. Can you please test it?

tty74it, 2018-05-13:
Yes now it works after #830  fix

Thank's
Tiziano

xoseperez, 2018-05-15:
Great. It will be released with 1.12.7

closed 2018-05-15 by xoseperez