xoseperez/espurna #839: Web and Arduino OTA Fails

darshkpatel, 2018-05-03:
I've been trying to use the OTA Feature in the Espurna Firmware since a couple of days , but i havent been able to get it working yet. 
Whenever i try the Arduino OTA , it uploads till 98% and then throws a error and prompts for password again.
[image]

Doing the WEB OTA , the firmware uploads till 40-50% and the ESP restarts and the upload stops. 

My GENERIC_8CH Firmware's size is around 507 kB while the ESP shows there's 2637824 bytes free. 

trying the two step updates, the firmware is successfully flashed through web ota but , after flashing i cannot find the esp on my network nor can i find a AP created by it . 

Any help troubleshooting this would be really helpful.

ColinShorts, 2018-05-07:
Perhaps you could grab some debug logs over serial, or are you able to disable some features to get the firmware below 500kB to see if ota will work?
Are you able to flash the full firmware via platformio?
Which esp module are you using, could you share the config to allow others to test?

darshkpatel, 2018-05-08:
Here's my arduino.h config , 
```
hardware
#define GENERIC_8CH

 Features
#define ALEXA_SUPPORT          1
//#define BROKER_SUPPORT         0
//#define DEBUG_SERIAL_SUPPORT   0
//#define DEBUG_TELNET_SUPPORT   0
//#define DEBUG_UDP_SUPPORT      1
#define DOMOTICZ_SUPPORT       0
#define HOMEASSISTANT_SUPPORT  1
#define I2C_SUPPORT            0
#define INFLUXDB_SUPPORT       0
//#define IR_SUPPORT             1
#define LLMNR_SUPPORT          0  // Only with Arduino Core 2.4.0
//#define MDNS_SERVER_SUPPORT    0
//#define MDNS_CLIENT_SUPPORT    1
#define MQTT_SUPPORT           1
#define NETBIOS_SUPPORT        0  // Only with Arduino Core 2.4.0
#define NOFUSS_SUPPORT         0
//#define NTP_SUPPORT            0
#define RF_SUPPORT             0
#define SCHEDULER_SUPPORT      0
//#define SPIFFS_SUPPORT         1
#define SSDP_SUPPORT           0
#define TELNET_SUPPORT         1
//#define TERMINAL_SUPPORT       0
#define THINGSPEAK_SUPPORT     0
//#define UART_MQTT_SUPPORT      1
#define WEB_SUPPORT            1

none of the sensors are enabled
```
with these settings i had a firmware compiled to the size of 469Kb which still resulted into the upload being unsuccessful with the module resetting itself whilst the firmware is uploading. 
I haven't tried platformio yet , i'll get back after trying that
I'm using a ESP12 Module, I've attached the image i found on the sellers page
[image]

darshkpatel, 2018-05-09:
> It will be useful to have the debug log of the device while you are trying
> to OTA flash it. Just telnet to it and copy paste the log here.

Trying to telnet the ESP results into a restart

darshkpatel, 2018-05-09:
I'd Hook it up with a serial monitor when i remove it from the enclosure next time. marking this issue closed for now as i was able to update through platformio.

closed 2018-05-09 by darshkpatel