xoseperez/espurna #840: Compile error class NTPClient' has no member named 'setNTPTimeout'

skyynet, 2018-05-06:
Sorry to bother but my libs don't always update. After installing 1.2.6 after having successfully compiled 1.2.4 I now face this error

Arduino: 1.8.3 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DIO, 1M (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Users/ingo/espurna/code/espurna/ntp.ino: In function 'void _ntpStart()':
ntp:48: error: 'class NTPClient' has no member named 'setNTPTimeout'
     NTP.setNTPTimeout(NTP_TIMEOUT);
         ^
/Users/ingo/espurna/code/espurna/ntp.ino: In function 'void _ntpConfigure()':
ntp:79: error: 'class NTPClient' has no member named 'setDSTZone'
     NTP.setDSTZone(dst_region);
         ^
exit status 1
'class NTPClient' has no member named 'setNTPTimeout'

I assume there's some updated lib which is not installed. Tried to update every lib which was mentioned and which sounded like ntp but didn't succeed so far. Any hin? Thanks.

xoseperez, 2018-05-06:
Please notice ESPurna uses a fork of the NtpClientLib at the moment. Check the [Arduino IDE wiki page](https://github.com/xoseperez/espurna/wiki/ArduinoIDE) to update the library to the required one.

skyynet, 2018-05-06:
Thanks. That fixed the issue. Didn't notice that. Maybe add a not to the changelog.

closed 2018-05-06 by skyynet