xoseperez/espurna #841: PWM fan controlling - frequency

evolucja, 2018-05-06:
Hello,
any chance of controlling the speed of 4 PIN computer fan directly from ESP8266 (Wemos D1)?

According to the PWM specification, the signal should have:

> PWM Frequency: Target frequency 25 kHz, acceptable operational range 21 kHz to 28 kHz

Using those settings:
`    // Info
    #define RELAY_PROVIDER      RELAY_PROVIDER_LIGHT
    #define LIGHT_PROVIDER      LIGHT_PROVIDER_DIMMER
    #define DUMMY_RELAY_COUNT   1

    // Light
    #define LIGHT_CHANNELS      1
    #define LIGHT_CH1_PIN       13
    #define LIGHT_CH1_INVERSE   0
    //`

the controlling works however a FAN clicking noise is noticable (I assume it's from the too low frequency).

Thanks
Dave

lobradov, 2018-05-07:
Espurna uses alternative PWM library that runs PWM at 19.6 kHz (if I did the math right). Normal ESP8266/Arduino runs PWM at 1kHz.

You can try changing frequency to 25kHz and see how your ESPurna behaves. You can do that by changing `LIGHT_MAX_PWM` parameter to 200 (either in `custom.h` or in `general.h`).

Note that higher PWM frequencies are not accurate (higher you go, less accurate it is), which is normally not a problem for LEDs and stuff, but might be an issue for motors or other "precise" things that depend on accurate timings. On this, both implementation (stock Arduino and alternative one) note that PWM timings are only accurate up to 3-4kHz, so 19kHz is already pretty inaccurate.

Another thing to note is that high frequencies will keep ESP CPU more occupied, so you might see less responsive interfaces (web, MQTT or whatever else you are using). Don't think that's going to be the case with moving from 19kHz to 25kHz, but it might start showing if you go even higher.

Good luck and let us know how it worked.

lobradov, 2018-05-07:
... taking a step back, you also might want to explore using specialized fan PWM controllers that can be controller from ESP8266 via I2C/Serial, like MAX31790 (made for controlling fans, I2C interface, few dollars a piece) or even separate ATtiny85 that will act as I2C slave.

Drivers don't exist in ESPurna yet, but hey - you are more than welcome to write and contribute them :)

evolucja, 2018-05-07:
Thank you for your very datailed answer.

Could you please just describe how to understand the following setting?

```
#if LIGHT_PROVIDER == LIGHT_PROVIDER_DIMMER
#define LIGHT_MAX_PWM           10000        // 5000 * 200ns => 1 kHz
#endif
```

Isn't the 10000 = 10kHz? How did you get the 19,6kHz?

lobradov, 2018-05-07:
erm, right, I was looking at line above that defined LIGHT_MAX_PWM to 255, which is 19.6kHz.
10000 is 500Hz.

So, math is: 
- 1s = 1,000,000,000 ns
- one "duty cycle pulse" is 200ns (fixed). 
- if you have 255 pulses in a duty cycle, it means you are running at 19607 cycles/s, aka 19.6 kHz.
- if you have 10,000 pulses, it means you are running at 500 Hz.

I'll go and change above comments, for future references.

evolucja, 2018-05-07:
Everything's clear now. I'll try changing those values and will report result.

lobradov, 2018-05-10:
Tried it? Is it working? 

Please update and close the issue if it's all good.

closed 2018-08-20 by stale[bot]

DIYglenn, 2018-08-27:
Does this work?
AFAIK the PWM signal is 5V, while the ESP8266 devices are 3.3V.
How about tach? I think it's pulled up to fan voltage (12V).