xoseperez/espurna #842: Espurna failing fixed IP 192.168.1.159

AlbertWeterings, 2018-05-06:
Can anyone do me a favor and put a espurna flashed device on fixed IP 192.168.1.159
With multiple devices Sonoff basic I noticed : 
- not able to get time by NTP
- 8 out of 10 times OTA fails result not functioning device only flashable via serial port
- intermittent webinterface not reachable 
- normal functioning over MQTT

I was having a sonoff basic in my network on this IP already for a long time. It was always a struggle to get new versions of Espurna loaded so this weekend I decided to replace the device with a new one.  From the first minute on it was not working well so I gave it a other IP that solved the issues so I took the old unit and gave it a new IP this device was also cured from the issues.  So I now want to know if it is device/firmware related or something in my network.

mcspr, 2018-05-07:
My bet is networking configuration (or some router quirks), if MQTT works somehow.

I am not seeing any issues using static ip assignment, 192.168.1.159 device running right now.

ColinShorts, 2018-05-07:
Networking seems the likely candidate, what is the ip address of your pc/laptop/whatever that you are flashing from? What about the mqtt server? If mqtt is lower and your flasher is higher then it could be the subnet mask that is misconfigured, but that seems unlikely. On the other hand, it could be some crazy issue caused by incorrect MTU size, or other strangeness. Can you reproduce the issue with other static ip's? Is it possible that there is another device with the same address?

AlbertWeterings, 2018-05-08:
Hi, thanks for your response. I'm still searching the cause. It happens only on IP 192.168.1.159 all other IP's are working fine. This network segment has 183 static IP's in use without any problems.

wildwiz, 2018-05-08:
My suspect is that you have another device lingering around on same IP address, that caused weird behavior to me in past and your description fits.

closed 2018-08-20 by stale[bot]