xoseperez/espurna #844: Button modes not working?

Fietspomp86, 2018-05-08:
I have a Lolin with Espurna, I'm trying to use a magnetic switch to determine if the door is open or closed.
But instead of just status 1 or 2 I also get status 3/4/5, as in long click etc.

This is my config and it seems right in my opinion?

```
#define BUTTON2_PIN         1
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON2_PRESS           BUTTON_MODE_ON
    #define BUTTON2_CLICK           BUTTON_MODE_OFF
    #define BUTTON2_DBLCLICK        BUTTON_MODE_NONE
    #define BUTTON2_LNGCLICK        BUTTON_MODE_NONE
    #define BUTTON2_LNGLNGCLICK     BUTTON_MODE_NONE
		#define BUTTON3_PIN         3
    #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON3_PRESS           BUTTON_MODE_ON
    #define BUTTON3_CLICK           BUTTON_MODE_OFF
    #define BUTTON3_DBLCLICK        BUTTON_MODE_NONE
    #define BUTTON3_LNGCLICK        BUTTON_MODE_NONE
    #define BUTTON3_LNGLNGCLICK     BUTTON_MODE_NONE
```

This is what I see on the Telnet output:
```
[071150] [BUTTON] Button #2 event 1
[071151] [MQTT] Sending ESPURNA_1C2F02/button/2 => 1 (PID 1)
[072278] [BUTTON] Button #2 event 4
[072279] [MQTT] Sending ESPURNA_1C2F02/button/2 => 4 (PID 1)
[073151] [BUTTON] Button #2 event 1
[073153] [MQTT] Sending ESPURNA_1C2F02/button/2 => 1 (PID 1)
[073661] [BUTTON] Button #2 event 2
[073663] [MQTT] Sending ESPURNA_1C2F02/button/2 => 2 (PID 1)
[075215] [BUTTON] Button #2 event 1
[075217] [MQTT] Sending ESPURNA_1C2F02/button/2 => 1 (PID 1)
[075439] [BUTTON] Button #2 event 1
[075440] [MQTT] Sending ESPURNA_1C2F02/button/2 => 1 (PID 1)
[075948] [BUTTON] Button #2 event 3
[075950] [MQTT] Sending ESPURNA_1C2F02/button/2 => 3 (PID 1)
[076224] [BUTTON] Button #2 event 1
[076226] [MQTT] Sending ESPURNA_1C2F02/button/2 => 1 (PID 1)
[076591] [BUTTON] Button #2 event 1
[076593] [MQTT] Sending ESPURNA_1C2F02/button/2 => 1 (PID 1)
[080071] [BUTTON] Button #2 event 3
[080072] [MQTT] Sending ESPURNA_1C2F02/button/2 => 3 (PID 1)
[089941] [BUTTON] Button #1 event 1
[089942] [MQTT] Sending ESPURNA_1C2F02/button/1 => 1 (PID 1)
[091860] [BUTTON] Button #1 event 4
[091862] [MQTT] Sending ESPURNA_1C2F02/button/1 => 4 (PID 1)
[092511] [BUTTON] Button #1 event 1
[092513] [MQTT] Sending ESPURNA_1C2F02/button/1 => 1 (PID 1)
[093013] [BUTTON] Button #1 event 2
[093014] [MQTT] Sending ESPURNA_1C2F02/button/1 => 2 (PID 1)
[093432] [BUTTON] Button #1 event 1
[093433] [MQTT] Sending ESPURNA_1C2F02/button/1 => 1 (PID 1)
[096152] [BUTTON] Button #1 event 4
[096153] [MQTT] Sending ESPURNA_1C2F02/button/1 => 4 (PID 1)
```

Also, Button numbers don't match the ones in the config.
Button 2 in config is button 1 in TELNET and so on.

xoseperez, 2018-05-08:
Button events are always triggered and reported via MQTT, even if they are not linked to any action.
Also, there is a refactor in the works to rebase all items, they should always be 0-based. Only place where they are still 1-based is in the hardware configuration.

Fietspomp86, 2018-05-11:
Allright, so basically it's not possible to make a simple open/close switch this way?

lobradov, 2018-05-11:
For real switch, you probably need to configured it like this: 
`#define BUTTON2_MODE BUTTON_SWITCH`

As documented here: 
https://github.com/xoseperez/espurna/wiki/Buttons-and-switches#buttons-and-switches

Fietspomp86, 2018-05-11:
Tried that as well, but then the only event is 2, (released). In this way there's no way to tell if the door is open or closed. This is also what is says in the Wiki:

> connected button is actually a flip-switch, only reports the click event on both transitions.

>

xoseperez, 2018-05-13:
Please notice that BUTTON_EVENT_RELEASED and BUTTON_EVENT_CLICK are actually the same event number. They are aliases for the same action but different kind of buttons (push buttons or toggle switches). So you are indeed receiving a BUTTON_EVENT_CLICK which is the only event a toggle switch can emit.

Fietspomp86, 2018-05-13:
Exactly, that's what I meant. This way there's no way to tell if the door is open or closed since the event is the same for both states.

xoseperez, 2018-05-13:
Right, that's the thing with toggle switches. It is not possible to sync the physical button with the state of the relay, since you can also change the relay using MQTT, REST, the web interface,...

untergeek, 2018-08-02:
But these are fixed position magnetic (reed) switches.  When there's a magnet in front of it, it's _always_ closed.  When the magnet moves away, it's _always_ open.  It seems like it should be possible to just show the state of whether the GPIO pin is high or low.

untergeek, 2018-08-02:
I love what this project is doing, but for my garage door opener, I'm going to have to go custom with micro python.  I don't feel comfortable enough in C to make all of the code changes necessary to add in actual switch behavior (not buttons).  I found out that buttons don't "retain" in MQTT, which is another dealbreaker for monitoring the status of a garage door.  

I do hope that proper support for reed switches is added at some point, whether it's an extension of button, or becomes a binary_sensor component type, like Home Assistant uses.

xoseperez, 2018-08-04:
If you can check the status of the switch you should use a pushbutton instead with a configuration like this:

```
    #define BUTTON2_PIN         1
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON2_PRESS           BUTTON_MODE_ON
    #define BUTTON2_CLICK           BUTTON_MODE_OFF
    #define BUTTON2_DBLCLICK        BUTTON_MODE_NONE
    #define BUTTON2_LNGCLICK        BUTTON_MODE_NONE
    #define BUTTON2_LNGLNGCLICK     BUTTON_MODE_NONE
```
But mind that the issue is still the same, the firmware cannot "move" to toggle switch to match an MQTT toggle message (for instance).

Unfortunately, not everything is available in a soft configuration setting yet. Changing the retain for buttons requires changing the second `false` to `true` here:

https://github.com/xoseperez/espurna/blob/f18c901c3f0f99211175c08a5b204311609f4447/code/espurna/button.ino#L32

untergeek, 2018-08-09:
I just want to make a combination device type that can do what Home Assistant does with a "cover," like a Garage Door.

This requires a relay (to trigger the garage door), and a binary_sensor, which is typically a reed switch that shows whether the door is open or closed.  MQTT monitor for relay triggering is already well supported by Espurna.  All that is really needed here is a way to monitor whether a given pin is HIGH or LOW, with a debounce timer on it.  There are no button presses, though you could still monitor the pin on the rising and falling rather than just high/low.  

It sounds like this requires a new device type, since unlike a button, it monitors the high/low state continuously, rather than just briefly, as with a button press.  If I get time, I'll brush up on my C and see if I can't write something to handle it.

xoseperez, 2018-08-13:
Maybe the EventSensor can do the job. It monitors any GPIO and can be configured to be event-based, so triggering an MQTT message when the states changes.

untergeek, 2018-08-14:
I will look into that.  Thank you.

untergeek, 2018-08-14:
The last piece of that puzzle would be to have Home Assistant MQTT auto-detection recognize this as a "cover," rather than as a separate switch and binary_sensor.

The discovery topic looks like:

```
<discovery_prefix>/<component>/[<node_id>/]<object_id>/<>
```
Where `<component>` is `cover`.  The JSON object then needs to include:
```
 {
  "name: NAME,
  "state_topic": "<discovery_prefix>/binary_sensor/[<node_id>/]<object_id>/<>",
  "state_open": ON_VALUE,
  "state_closed": OFF_VALUE,
  "command_topic": "<discovery_prefix>/cover/[<node_id>/]<object_id>/set"
}
```
I'm not sure how you'd add this to your code, but I like the idea of having a "cover" type for Home Assistant.

closed 2018-10-20 by stale[bot]