xoseperez/espurna #846: Google Home support

hyteoo, 2018-05-08:
With Google Home getting more and more traction, what would it take to also add support for that alongside Alexa?

ColinShorts, 2018-05-08:
Quite a bit of work. Alexa support isn't really Alexa support but an emulation of Phillips hue local discovery and control. Google home needs to connect to an online service to control or interact with devices. If that is something that you need then you might want to look at using openHAB or similar as it would be easy enough to setup in a few hours, or a day or so by the time you read through the docs 😀

hyteoo, 2018-05-08:
Thanks for the quick answer
I'm currently testing away integrating it for voice commands, however it's currently a GoogleHome - IFTTT - NodeRed
Command processing is done in NodeRed so not really taking advantage of G. Assistant's contextual knowledge.
If Google Home would know what devices & where they are, it obviously be much integrated.
I know it would be a lot of work, maybe even needs a partnership with Google for that (as they do with all other devices). However I am just thinking that if they support MagicHome, then maybe Espurna would not be out of the question.
Maybe sometime in the future....who knows :)

Until then, keep up the great work!

ColinShorts, 2018-05-08:
earlier I said Phillips hue, but it is actually FauxmoESP that is used, sorry for any confusion... I was confusing myself with the phillips emulation that is available in openHAB.

Anyways, you probably guessed, but I use openHAB with the cloud connector in order to get google assistant working, it's not perfect by any means - it only currently handles switches, lights and to a lesser degree, thermostat devices. It's still quite early days, so I'm hoping it will improve in the coming months.

Unlike IFTTT, the google assistant openHAB cloud connector integration uses the openHAB api's to pull back all devices rather than those "exposed" through the cloud connector. What I am trying to get at is that there needs to be some infrastructure in place for google assistant to integrate, so it is considerably more work than adding a library or two. I've never gotten around to really trying the NodeRED route, but I found IFTTT to be a bit on the fiddly side for my liking.

If you've already got the devices setup with MQTT, adding openHAB to the mix would be fairly trivial and a pretty simple way of controlling basic devices via google assistant and/or Alexa if you so choose. You can still keep all of the back-end flows in NodeRED and/or IFTTT.

hyteoo, 2018-05-09:
I tried both OpenHab and Hass, however as much as I wanted to like them, they just are not there yet. Lots of quirks, if it's not V2 it's only v1, ...etc. 
UI - one of the most important topics in a home automation dashboard is awful.
Bringing devices together (especially devs running the same firmware) should be much simpler that that.

In nodered you can do almost anything related to the interface, works great on mobile and scales well.
Also it's easy to process webrequests from the outside.

IFTTT is a temp workaround, it's too limited to really do anything. Biggest problem is when you have 2-3 G Home in different rooms and you want to "turn on the lights". There is no source information.

@xoseperez Regarding your suggestion, I know that wiki wery well. Biggest issue there is that it works ok..ish for 1 device... I have around 18 espurna switches, sensors, custom boards currently all tied together with MQTT. Exposing each one to the WAN would be a logistical nightmare (each device requiring setup in Espurna, router/firewall, IFTTT).

One other option I am exploring is Google Actions that could provide more integration..

Nephiel, 2018-10-14:
How about this? https://github.com/i8beef/HomeAutio.Mqtt.GoogleHome

Nephiel, 2018-10-14:
Not yet, but I intend to. I'm already too deep into the Google ecosystem to back out. I'd rather not have every lamp in the house talking to third party cloud providers on top of that, if I can build and host my own service using open source.

Nephiel, 2018-10-25:
I've been testing, and found it requires adding some MQTT "glue". I'm using Node-RED:  

 - Google uses ranges 0-100 for brightness while ESPurna uses 0-255.  
 - RGB color range in Google is is a single integer instead of ESPurna's #RRGGBB or (R,G,B).  
 - It seems there is no UI yet to set color from the Google Home app - only brightness. It does work with voice though ("OK Google, set the attic lamp to blue").
 - Some colors by name are just wrong on Google's end, e.g. "rosa" (pink) returns the RGB value for white (#FFFFFF), and "blanco" (white) returns #FFDEAD.  

Still trying to figure out how to get the Report State feature working.

Nephiel, 2018-10-28:
I tried the Report State feature. It's supposed to reduce network traffic by storing the state of each device in HomeGraph, but each device must update it every time anything changes, to keep it up to date. Still needs work, and didn't really see any improvements during the short time I had it working. Ended up just leaving `willReportState` set to `false`.  

Now, as a result of the tests, I have multiple duplicate devices in Google Home that I seem unable to remove, even after unlinking. But that's an issue with the Google Home app.

TMaYaD, 2019-07-11:
With google releasing new a [Local home SDK](https://developers.google.com/actions/smarthome/reference/local/) it might be a good time to revisit this?

mcspr, 2019-07-11:
According to https://developers.google.com/actions/smarthome/concepts/local

> The local execution app contains the business logic to process the intents sent by the Local Home platform and access your smart device via the local area network. No modification of your hardware is expected to integrate local execution.
...
> In production, Google hosts your app in a secure JavaScript sandbox environment on the user’s Google Home speaker or Nest display.
...

I am under the impression that it needs some controller app first, which will both discover devices and dispatch network request to trigger actions (for example, using REST api calls)

TMaYaD, 2019-07-15:
Yeah. On further reading through the local home SDK, it seems like this needs to be a separate companion app that runs on google home and connects to espurna.