xoseperez/espurna #851: RF Bridge sends multiple RfKey clicks

law1964, 2018-05-10:
I have flashed 1.12.6 to my RF Bridge R2 V1.0. It is able to learn the key for the RF transmitter (raw code is 292C015404062DE832). The bridge is also able to transmit it. Unfortunately, it seems to send it multiple times. I can hear multiple clicks from the relay on the RF receiver.

My Switch settings are:
Boot Mode = Always OFF
Pulse mode = Normally OFF
Pulse time (s) = 0.1 <-- it doesn't matter. I've set this up to 0.5 seconds and it still sends multiple key clicks.

My RF settings are:
Switch ON = 292C015404062DE832
Switch OFF = (no code)

I've tested this with Tasmota too. Tasmota is able to reproduce the RfKey flawlessly. I hear only one click from the receiver relay. The json encapsulation from the Tasmota console is:

MQT:tele/sonoff/RESULT={"RfReceived":{"Sync":10560,"Low":340,"High":1030,"Data":"2DE832",RfKey":1}}

Please have a look and let me know how to fix this.

Jackson

lobradov, 2018-05-10:
You might want to check out `RF_SEND_TIMES` option, which is, by default, set to 4 retransmits. 
This is to cater for unreliable devices where multiple retransmits are often needed to make sure that message is actually delivered. 

You can change the option in `custom.h` (making sure that your firmware is build with `-DUSE_CUSTOM_H` option)

law1964, 2018-05-13:
@xoseperez Thank you for the response.

I will try platformio when I get the chance to install it and learn how to use it.

I removed the -DUSE_CUSTOM_H build flag to change the RF_SEND_TIMES back to 4. and re-flashed the RF Bridge.  I set Switch ON and Switch OFF both to 292C015404062DE832 and tested. No luck. I hear fewer clicks from the receiver for the blinds, but it still starts and stops when I activate the switch.

Just for testing purposes, I set the RF_SEND_TIMES to 2, added the -DUSE_CUSTOM_H build flag, and recompiled. Still no luck. The blinds relay still clicks multiple times with 1) Switch ON filled in alone and 2) Switch ON and OFF both filled in with the code.

I reverted back to RF_SEND_TIMES = 1 and it works!

Something strange to note: each time I was uploading the firmware back to the RF Bridge, it re-sent the RF code to the blind automatically (ie. the RF receiver for the blinds clicked without user initiation).

I'm back to RF_SEND_TIMES=1, Switch ON = code, Switch OFF = no code. This is the only configuration that works every single time.

wildwiz, 2018-05-15:
unfortunately the RF_SEND_TIMES has value greater than one because other devices don't pick reliably a single burst of pulses ... when ON and OFF code are distinct that is not a problem and in fact it multiple send increase the reliability (on RX side, the rc-switch library used in RFB_DIRECT always ignores the first burst of pulses ... go figure)

xoseperez, 2018-05-15:
If the codes for ON and OFF are the same it should only send it once to avoid this problem. If it's still sending it several times then there is a bug.

xoseperez, 2018-05-22:
Tested on two different versions of the RFBridge and it works OK. If the code for ON and OFF is the same the message is sent only once. Mind it has to be exactly the same. Do not learn both ON and OFF codes since each learn process might produce a slightly different code, that's normal. Instead just learn the ON and copy paste the ON value to the OFF text box and click save to store it.

law1964, 2018-05-22:
Thank you for testing this.

I downloaded and installed the espurna-1.12.6-itead-sonoff-rfbridge.bin (I had deleted the non-working builds from arduinoIDE) from github and tested again, making certain that the both the ON and OFF codes were exactly the same. This still does not work for me. I am still getting multiple RF sends. 

I re-installed my build from Espurna with the RF_SEND_TIMES=1 and it works properly every time.

You can't fix something that you can't reproduce. Therefore, if I am the only one experiencing this issue, please close this bug report. I will continue to use the RF_SEND_TIMES=1 in a custom build.

xoseperez, 2018-05-23:
@law1964 Can you open a debug window to the device via telnet and copy paste here the log while it sends the messages?

law1964, 2018-05-23:
These are the telnet debug messages for rfbridge firmware downloaded from github with ON and OFF set to 2918015403FC2DE832:

[155245] [TELNET] Client #0 connected
[169424] [WEBSOCKET] Requested action: relay
[169427] [RELAY] #0 scheduled ON in 0 ms
[169431] [RELAY] #0 set to ON
[169434] [RFBRIDGE] Enqueuing MESSAGE '2918015403FC2DE832' 1 time(s)
[169737] [RELAY] #0 scheduled OFF in 0 ms
[169740] [RELAY] #0 set to OFF
[169743] [RFBRIDGE] Enqueuing MESSAGE '2918015403FC2DE832' 1 time(s)
[170279] [RFBRIDGE] Action 0xA0
[170745] [RELAY] Saving mask: 0
[170782] [RFBRIDGE] Action 0xA0

The first time, I was able to activate the switch, but I needed to reboot the device from the main menu to be able to activate the switch another time. I repeated this test several times. The debug messages are all the same. Each time I activated the switch, it sent the RF signal two times.

For ON = 2918015403FC2DE832 and OFF = NULL, the signal was send 4 times, as expected because RF_SEND_TIMES defaults to 4 (see below):

[565331] [TELNET] Client #0 connected
[573643] [WEBSOCKET] Requested action: relay
[573647] [RELAY] #0 scheduled ON in 0 ms
[573649] [RELAY] #0 set to ON
[573652] [RFBRIDGE] Enqueuing MESSAGE '2918015403FC2DE832' 4 time(s)
[573955] [RELAY] #0 scheduled OFF in 0 ms
[573958] [RELAY] #0 set to OFF
[574507] [RFBRIDGE] Action 0xA0
[574960] [RELAY] Saving mask: 0
[575498] [RFBRIDGE] Action 0xA0
[575998] [RFBRIDGE] Action 0xA0

For my customized build with RF_SEND_TIMES=1 and ON=2918015403FC2DE832 and OFF=NULL, the signal was sent only one time (see below):

[115287] [TELNET] Client #0 connected
[115683] [WEBSERVER] Request: GET /index.html
[116034] [WEBSOCKET] #1 connected, ip: 192.168.20.21, url: /ws
[125887] [WEBSOCKET] Requested action: relay
[125891] [RELAY] #0 scheduled ON in 0 ms
[125893] [RELAY] #0 set to ON
[125896] [RFBRIDGE] Enqueuing MESSAGE '2918015403FC2DE832' 1 time(s)
[126199] [RELAY] #0 scheduled OFF in 0 ms
[126202] [RELAY] #0 set to OFF
[126743] [RFBRIDGE] Action 0xA0
[127205] [RELAY] Saving mask: 0

xoseperez, 2018-05-23:
Ah. You have pulse mode enabled, right? The first log is compatible with this. You manually turn it on, it  enqueues the message once and then the pulse module turns it off again and enqueues the message a second time... just disable pulse ("Don't pulse") and you will have a normal behaviour.

law1964, 2018-05-23:
It works! I am glad you were able to figure it out. Unfortunately, now the switch representation is latched which is meaningless for an application that uses a momentary switch.

law1964, 2018-05-23:
I meant "toggled" not "latched".

xoseperez, 2018-05-25:
Yeah, there was a requests somewhere about having the option to use a simple button image in the web interface instead of switches when codes for ON and OFF are the same

xoseperez, 2018-06-04:
See #758, point 8.

closed 2018-06-04 by xoseperez