xoseperez/espurna #852: TYWE3S Support

sandman72, 2018-05-10:
I had some time today to play around with an EStink Wifi Smart plug box ( https://amzn.to/2I9477J ). It's based on the TYW3S wifi controller modules that have an ESP8266 chip in them. 

These work just like generic ESP8266 1M boards, but I had to manually reset it just before uploading. Here is the TYWE3S module pinout for reference:

[image]

Here is a hardware definition for it if anyone is interested: 
   
    // ESTINK 4 USB ports plus 3 Relay
    #elif defined(ESTINK_WIFI_RELAY)

    // Info
    #define MANUFACTURER        "ESTINK"
    #define DEVICE              "WIFI_RELAY"

    // Buttons
    #define BUTTON1_PIN         16
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    //    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          14  // USB power
    #define RELAY2_PIN          13  // power plug 1
    #define RELAY3_PIN          4   // power plug 2
    #define RELAY4_PIN          15  // power plug 3
	
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL
    #define RELAY3_TYPE         RELAY_TYPE_NORMAL
    #define RELAY4_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            0   // power led
    #define LED2_PIN            12  // power plug 1
    #define LED3_PIN            3	// power plug 2
    #define LED4_PIN            5   // power plug 3

    #define LED1_PIN_INVERSE    1
    #define LED2_PIN_INVERSE    1
    #define LED3_PIN_INVERSE    1
    #define LED4_PIN_INVERSE    1
	
    #define LED2_MODE	LED_MODE_FOLLOW
    #define LED3_MODE	LED_MODE_FOLLOW
    #define LED4_MODE	LED_MODE_FOLLOW
	
    #define LED2_RELAY	2
    #define LED3_RELAY	3
    #define LED4_RELAY	4

xoseperez, 2018-05-13:
Nice. Do you want to do a PR with your configuration?

sandman72, 2018-05-13:
I haven't setup a git repo for it. It might be best/easiest if you just add it to the configuration.h

xoseperez, 2018-05-15:
No problem. I can add it myself. Thank you!

xoseperez, 2018-05-22:
Staged for release

closed 2018-05-22 by xoseperez