xoseperez/espurna #854: Sonoff RF Bridge Direct Hack not working

derFrickler, 2018-05-11:
Thanks for the direct mod, as i have some Remotes that did not work with the original one. 

I did the hardware mod as described on https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack and flashed the firmware image "espurna-1.12.6-itead-sonoff-rfbridge-direct.bin" but the RF Bridge is unable to detect/learn any command. 
Without the mod and with the non direct firmware it works with my remote. although my unmodified bridge receives the commands. 

Any Ideas here?

wildwiz, 2018-05-11:
Hi, are you able to send a picture of the modified board and some details about the remotes ? 
We may either have an issue with the ESP8285 not receiving the data or an issue with rc-switch library not decoding your remotes.

derFrickler, 2018-05-11:
Wiring of the modified board is fine, as shown in the wiki. tried with 470Ohm resistors and directly wired.
Remote is an Intertechno, which works fine with the unmodified bridge. Tried other remotes as well that did not work with the unmodified one, none of them is received. learing in the web gui is not working and MQTT rfout is not sending anything.

wildwiz, 2018-05-11:
I got a moment to put the direct firmware on a R2 V1.0 bridge that I received recently, and I experience your same issue, while the same remote is working fine with a (R1)V2.0. Seems that GPIO4 is held low by the processor even if it is instructed otherwise. Investigating it.

wildwiz, 2018-05-12:
Seems that the R2V1.0 version of sonoff bridge has a routing error on the PCB. The actual lines for GPIO4 and GPIO5 are swapped, while the serigraphy has not been updated. This obviously prevented the RF signal from getting where the software expects it (same for transimt mode of course).

I will update the wiki soon with more pictures and the details that follow up here.
Meanwhile, there are two ways to deal this difference between the boards:

1. Invert the connections for GPIO4 and GPIO5 (recommended, if you feel comfortable).
2. Change the definition in platformio.in (no more hardware changes, but this is the only advantage). This will produce firmwares that only work on the R2V1.0 but no longer on other versions of sonoff bridge. It can become tricky if you have more than one with mixed versions, especially as there is no way to verify which board is which but opening them.

The change to be applied adds -DRFB_RX_PIN=5 -DRFB_TX_PIN=4 to the build flags and the diff looks as this:
```
git diff
diff --git a/code/platformio.ini b/code/platformio.ini
index 54150a1..4438b9e 100644
--- a/code/platformio.ini
+++ b/code/platformio.ini
@@ -750,7 +750,7 @@ board = esp01_1m
 board_flash_mode = dout
 lib_deps = ${common.lib_deps}
 lib_ignore = ${common.lib_ignore}
-build_flags = ${common.build_flags_1m} -DITEAD_SONOFF_RFBRIDGE -DRFB_DIRECT
+build_flags = ${common.build_flags_1m} -DITEAD_SONOFF_RFBRIDGE -DRFB_DIRECT -DRFB_RX_PIN=5 -DRFB_TX_PIN=4
 monitor_baud = 19200
 extra_scripts = ${common.extra_scripts}

@@ -761,7 +761,7 @@ board = esp01_1m
 board_flash_mode = dout
 lib_deps = ${common.lib_deps}
 lib_ignore = ${common.lib_ignore}
-build_flags = ${common.build_flags_1m} -DITEAD_SONOFF_RFBRIDGE -DRFB_DIRECT
+build_flags = ${common.build_flags_1m} -DITEAD_SONOFF_RFBRIDGE -DRFB_DIRECT -DRFB_RX_PIN=5 -DRFB_TX_PIN=4
 upload_speed = 115200
 upload_port = "${env.ESPURNA_IP}"
 upload_flags = --auth=${env.ESPURNA_AUTH} --port 8266
```

wildwiz, 2018-05-12:
Another note: the RF codes produced by the "direct" processing are different from the ones produced by the "vanilla" bridge for same remote. This was a design choice, due to the different kind of information passed by the hardware processor and the software library. This should not be a problem given that, well, you modify the hardware to switch from one configuration to the other.

xoseperez, 2018-05-13:
Closing this
@wildwiz has updated the wiki and I have just uploaded his latest images to hack v2.0 boards.

closed 2018-05-13 by xoseperez

derFrickler, 2018-05-13:
Perfect, the mod works now! Thanks a lot!

...unfortunately my other remote (http://www.rolleaseacmeda.com/docs/default-source/Instructions/rf200-prog-inst-wirefree-sml_med-motor_201306.pdf) is still not detected. Seems to use some strange encoding...

wildwiz, 2018-05-22:
It would be great if you could send me a sampled waveform or the corresponding pulse timings. As you have already modified the bridge, there is a pending PR I have a pending PR that can put the modified bridge in RF trace mode and extract the timings directly from its radio. You can find it here https://github.com/wildwiz/espurna/tree/on-rx-dump and the documentation is here #854