xoseperez/espurna #861: RF Bridge does not learn

henfri, 2018-05-13:
Hello,

thanks for making espurna!
I have a seav TXS 4 NEW. Here the datasheet:
http://www.seav.it/sites/default/files/prodotti/schede_tecniche/Txx%204%20GB.pdf

From the spec:

>  Mod. TXS 4: 433,92 MHz
> 4 channel transmitter with the capability for 12 and 18 bit
> code transmission and the division of codes for common
> and personalised command management.

Unfortunately, espurna does not react when I try to learn the codes (I get the beep, but that's it)
Is there anything I can do?

Regards,
Hendrik

wildwiz, 2018-05-13:
I see you have already reached the RF_RAW_SUPPORT page https://github.com/Portisch/RF-Bridge-EFM8BB1 page for modifying the EFM8BB1 firmware. 

If you didn't already, you might have a look at the RFB_DIRECT page https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack, it should be able to support all remotes supported by the rc-switch library https://github.com/sui77/rc-switch and for your description I think (no guarantees) you have good chances for it to work.

henfri, 2018-05-13:
Hello,

thanks for your hint. I was not aware of this hack.
I am not sure my soldering skills are up to this task...

Regarding https://github.com/Portisch/RF-Bridge-EFM8BB1 : 
Yes, I found this after just my post here.
I wonder why Itead did opt for the EFM8BB1 instead of this?

Greetings,
Hendrik

wildwiz, 2018-05-13:
I can guess a couple of reasons. 
* (a) if I remember properly, they already used the same microcontroller to decode RF in another product, so possibly it was just matter to drop in rather than develop the code and device from scratch.
* (b) they might be worried that the ESP8266, handling the WI-FI which is a complex beast, could give issue when combining with tight timing code for decoding. 

Actually, it is very possible that sometimes the ESP8266 driven processing may be disturbed by the WI-FI, but most codes are repeated and the chance of a packet being lost on WI-FI already exist and is not incremented significantly by the software task.

nunofgs, 2018-06-06:
@wildwiz I have a similar issue. I cannot get my RF bridge to learn but I don't even get the buzzer sound when I click "Learn" (note: with the original Sonoff firmware, it buzzed just fine)

Today I tried flashing the regular espurna (non-direct) version. Then, when that didn't work, I fired up my soldering iron and performed the hack described [in the wiki](https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack).

I have board R1 V2.0 by the way. I performed the hack and flashed the "direct" version but, same result. The buzzer does not buzz and obviously, clicking my remote does not produce a code.

Any ideas?

wildwiz, 2018-06-07:
With the -direct hack the buzzer is not expected to buzz anymore (the RF led will still flash), as the microcontroller handling it has been bypassed. 

While the hack widens the capabilities of recognition, it is not universal. However, one nice thing that we can do with the modified board is capture pulse timings and study them ! If you want to give it a try, you can find more info here #864 

In case you do, please open a new issue with the info you gathered, pulses, brand of RC and whatever

closed 2018-08-20 by stale[bot]