xoseperez/espurna #865: Configuration For Hass not showing up in Sonoff POW

seetendra, 2018-05-16:
Hello Everone

Iam not sure if his is possible right now 
For the latest version of Espurna when we got to Hass in the WebUI there is option to click on configuration and you get the code that you need to enter in the config.yaml file in home assistant

This works with all the Sonoff basics no problem.
But it does not work for the Sonoff POW where we need to pull the power usage details to home assistant as well

Is this a Know issues? If yes can some one please help me with the how do i pull the info to the home assistant

seetendra, 2018-05-16:
I have also noticed that there is an option to use Use JSON payload in MQTT
If i turn that on
Iam able to turn on the device using home assistant but unable to turn it off 

This is the code that iam using in one of the sonoff POW and the same is notice in Sonoff Basic as well

Sonoff POW

- platform: mqtt
    name: "Fridge"
    state_topic: Fridge/relay/0
    command_topic: Fridge/relay/0/set
    payload_on: "1"
    payload_off: "0"
    availability_topic: Fridge/status
    payload_available: 1
    payload_not_available: 0
    qos: 1
    retain: true

Sonoff Basic

switch:
  - name: Dinning Room
    platform: mqtt
    state_topic: Dinning Room/relay/0
    command_topic: Dinning Room/relay/0/set
    payload_on: 1
    payload_off: 0
    availability_topic: Dinning Room/status
    payload_available: 1
    payload_not_available: 0

mcspr, 2018-05-16:
Should be fixed by https://github.com/xoseperez/espurna/pulls/850
JSON.parse breaks on `haConfig` string starting from quoted "A" unit of measurement.

For now you can use **Debug** panel and send `ha.config` command to see the config.

seetendra, 2018-05-16:
Hi Max

Thx a lot for your reply on this 
But it is giving me errors

 - Invalid config for [sensor.mqtt]: value is not allowed for dictionary value @ data['device_class']. Got 'sensor'. (See ?, line ?). Please check the docs at https://home-assistant.io/components/sensor.mqtt/
    - platform: mqtt
      device_class: sensor
      name: Fridge energy
      state_topic: Fridge/energy
      unit_of_measurement: kWh

seetendra, 2018-05-16:
@mcspr 

This is what i entered

sensor:
  - name: Fridge current
    platform: mqtt
    device_class: sensor
    state_topic: Fridge/current
   unit_of_measurement: "A"

mcspr, 2018-05-16:
Huh. That's too fixed by #859 :) Just remove every line with device_class

xoseperez, 2018-05-18:
@seetendra Can you confirm that @mcspr pull requests fix your issues? They are both merged in the dev branch.

seetendra, 2018-05-18:
Hi @xoseperez  Yes sorry forgot to reply it did fix the issues

seetendra, 2018-05-18:
I just has to get the device_class removed and also enter ha.config command in Dubug to get the details
After that it worked for me.

xoseperez, 2018-05-19:
Great. Will be released with next version.

closed 2018-05-19 by xoseperez