xoseperez/espurna #867: SonOff POW R2 using CSE7759B not reporting current, active power & energy

ServerCare, 2018-05-16:
I have a freshly received POW R2 (rev v1.0 nov 30 2017) that has a CSE7759B measuring chip in place. The status page shows me 234 V (which could be OK) but all other values are zero.

Is my device broken, is my chip not a regular one (as I see CSE7766 mentioned) or is the firmware not reading the values?

What can I do to resolve this?

jeanpaul79, 2018-05-16:
Hello,

I have the same problem on a Sonoff Pow since 1.12.5. If you turn off the relay it should display the voltage.
I think there is a regression problem in 1.12.5 due to "Force reporting power values as 0 if relay is off"

Jean-Paul

cflurin, 2018-05-17:
Hi,
The same problem with a Sonoff Pow R2 V1.0, CSE7766 (ESPUrna v1.12.6)

gn0st1c, 2018-05-17:
@xoseperez : may be espsoftwareserial issue?

xoseperez, 2018-05-18:
Mmm... I think there are three different issues here:
* Issue with POW R2 with CSE7759B according to @ServerCare 
* Issue with POW R2 with CSE7766 according to @cflurin 
* Issue with POW "R1" since 1.12.5 according to @jeanpaul79 

Actually the CSE7759B chip is not supported (yet, but the other two should work fine. And I have two "R1" with 1.12.6 working fine. @jeanpaul79 do you have that setting active (the "Force reporting power values as 0 if relay is off"?

jeanpaul79, 2018-05-18:
Hello,

I'm using the binary available on github. If relay is on, everything is at 0, even voltage. If relay is off then voltage is at 220V (but power is at 0 but that's normal because relay is off).

I don't find the setting you are talking about?

Regards

xoseperez, 2018-05-21:
I have compared the CSE7766 and CSE7759B datasheets and they are very similar. The CSE7759B is less accurate (twice as much less accurate) and it has special reset requirements and a coefficient loaded flag. But none of the difference has given me a clue on why it does not read the values.

My POW R2 has a CSE7766, so I cannot test...

@ServerCare would you mind update to the latest version in the dev branch and enable SENSOR_DEBUG in the sernsors.h file? Then flash your device and monitor the log via telnet, copy-paste here a few rows of data. Let's see if we can figure out what's going on.

cflurin, 2018-05-22:
Update:
I tried Tasmota v5.14.0. It seems to work with my Sonoff Pow R2.
[image]

cflurin, 2018-05-23:
Hi,
Yes, I think the `CHIPSEA 7766-SO` is a `CSE7766`

[image]

I used the pre_build binary `espurna-1.12.6-itead-sonoff-pow-r2.bin` following the wiki:

```sh
esptool.py --port /dev/cu.usbserial-A50285BI erase_flash
esptool.py --port /dev/cu.usbserial-A50285BI write_flash --flash_mode dout 0x00000 espurna-1.12.6-itead-sonoff-pow-r2.bin
```

xoseperez, 2018-05-23:
@cflurin can you test building and flashing the latest version from the dev branch? I did some changes yesterday that could fix your issue...

ServerCare, 2018-05-23:
Trying to build the dev (first time, as I used to simply flash the binaries) but I keep bumping into: the error
class SoftwareSerial' has no member named 'enableIntTx'
in the Arduino IDE

ServerCare, 2018-05-23:
I did order a few more POW R2's,  so I hopefully should be able to check more once they arrive...

xoseperez, 2018-05-23:
ESPurna uses this version of ESPSoftwareSerial: https://github.com/plerup/espsoftwareserial#3.4.1

cflurin, 2018-05-23:
@xoseperez 
I'v compiled the latest dev version with PlatformIO but I had an issue.
It seems I can't enter any characters, e.g. changing the password doesn't work.
I tried a second time after I've changed general.h

```
#define WEB_FORCE_PASS_CHANGE       0
#define WS_AUTHENTICATION           0 
```
But I can't change the wifi settings.

cflurin, 2018-05-24:
The same here: https://github.com/xoseperez/espurna/issues/879

Anyway power measuring with the latest dev version still doesn't work.

ServerCare, 2018-05-24:
So... I compiled & flashed the R2 with the dev build.  Here's some debug output (from the debug window, as I wasn't going to put power on it while connected to terminal)

[013875] [WEBSOCKET] #1 connected, ip: 192.168.14.11, url: /ws
[013950] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 F7 00 3D 36 03 F5 02 4C 72 50 CA 41 F3 61 00 [014060] [SENSOR] CSE7766: _process: 55 5A [014164] [SENSOR] CSE7766: _process: 55 5A 02 B7 [014268] [SENSOR] CSE7766: _process: 55 [014371] [SENSOR] CSE7766: _process: 55 [014474] [SENSOR] CSE7766: _process: 55 5A 02 [014579] [SENSOR] CSE7766: _process: 55 5A 02 [014684] [SENSOR] CSE7766: _process: 55 5A 02 [014789] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 [014893] [SENSOR] CSE7766: _process: 55 [014997] [SENSOR] CSE7766: _process: [015101] [SENSOR] CSE7766: _process: 55 [015205] [SENSOR] CSE7766: _process: 55 [015310] [SENSOR] CSE7766: _process: [015422] [SENSOR] CSE7766: _process: 55 [015527] [SENSOR] CSE7766: _process: 55 [015631] [SENSOR] CSE7766: _process: 55 [015735] [SENSOR] CSE7766: _process: [015840] [SENSOR] CSE7766: _process: 55 [015944] [SENSOR] CSE7766: _process: 55 [016048] [SENSOR] CSE7766: _process: 55 [016152] [SENSOR] CSE7766: _process: 55 [016257] [SENSOR] CSE7766: _process: 55 [016359] [SENSOR] CSE7766: _process: [016463] [SENSOR] CSE7766: _process: 55 [016567] [SENSOR] CSE7766: _process: 55 [016671] [SENSOR] CSE7766: _process: 55 [016776] [SENSOR] CSE7766: _process: 55 [016890] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 F9 [016986] [SENSOR] CSE7766: _process: 55 [017099] [SENSOR] CSE7766: _process: [017194] [SENSOR] CSE7766: _process: 55 [017307] [SENSOR] CSE7766: _process: 55 [017409] [SENSOR] CSE7766: _process: F2 [017512] [SENSOR] CSE7766: _process: F2 [017617] [SENSOR] CSE7766: _process: F2 [017722] [SENSOR] CSE7766: _process: F2 [017824] [SENSOR] CSE7766: _process: F2 [017937] [SENSOR] CSE7766: _process: F2 F2 [018349] [SENSOR] CSE7766: _process: F2 [018454] [SENSOR] CSE7766: _process: F2 [018560] [SENSOR] CSE7766: _process: F2 [018665] [SENSOR] CSE7766: _process: F2 [018770] [SENSOR] CSE7766: _process: [018875] [SENSOR] CSE7766: _process: F2 [018981] [SENSOR] CSE7766: _process: F2 [019086] [SENSOR] CSE7766: _process: F2 [019192] [SENSOR] CSE7766: _process: F2 [019297] [SENSOR] CSE7766: _process: F2 [019501] [SENSOR] CSE7766: _process: [019614] [SENSOR] CSE7766: _process: F2 [019709] [SENSOR] CSE7766: _process: F2 [019822] [SENSOR] CSE7766: _process: F2 [019928] [SENSOR] CSE7766: _process: F2 [020031] [SENSOR] CSE7766: _process: F2 [020136] [SENSOR] CSE7766: _process: [020240] [SENSOR] CSE7766: _process: F2 [020345] [SENSOR] CSE7766: _process: F2 5A [020451] [SENSOR] CSE7766: _process: F2 [020556] [SENSOR] CSE7766: _process: F2 [020659] [SENSOR] CSE7766: _process: F2 [020763] [SENSOR] CSE7766: _process: F2 [020868] [SENSOR] CSE7766: _process: F2 [020973] [SENSOR] CSE7766: _process: F2 [021078] [SENSOR] CSE7766: _process: F2 [021181] [SENSOR] CSE7766: _process: F2 5A 02 B7 [021287] [SENSOR] CSE7766: _process: F2 [021392] [SENSOR] CSE7766: _process: F2 [021497] [SENSOR] CSE7766: _process: F2 [021602] [SENSOR] CSE7766: _process: F2 [021707] [SENSOR] CSE7766: _process: F2 [021811] [SENSOR] CSE7766: _process: F2 [021916] [SENSOR] CSE7766: _process: F2 [022021] [SENSOR] CSE7766: _process: F2 [022126] [SENSOR]

ServerCare, 2018-05-24:
that was with some load connected to it; again only voltage shows. (and joule seems to count up, even with no load)

Here's what I got when it was powered through my USB:
[337229] [WEBSOCKET] Requested action: relay
[337231] [RELAY] #0 scheduled ON in 0 ms
[337234] [RELAY] #0 set to ON
[338169] [SENSOR] CSE7766 @ HwSerial - current: 0.000A
[338172] [SENSOR] CSE7766 @ HwSerial - voltage: 0V
[338174] [SENSOR] CSE7766 @ HwSerial - power: 0W
[338175] [SENSOR] CSE7766 @ HwSerial - energy: 0J
[338236] [RELAY] Saving mask: 1
[341400] [WEBSOCKET] Requested action: relay
[341402] [RELAY] #0 scheduled OFF in 0 ms
[341405] [RELAY] #0 set to OFF
[342407] [RELAY] Saving mask: 0
[343793] [MAIN] Uptime: 343 seconds
[343795] [MAIN] Free heap: 18912 bytes
[343797] [MAIN] Power: 3518 mV
[343798] [MAIN] Time: 2018-05-24 13:58:05
[344179] [SENSOR] CSE7766 @ HwSerial - current: 0.000A
[344182] [SENSOR] CSE7766 @ HwSerial - voltage: 0V
[344183] [SENSOR] CSE7766 @ HwSerial - power: 0W
[344184] [SENSOR] CSE7766 @ HwSerial - energy: 0J
[350182] [SENSOR] CSE7766 @ HwSerial - current: 0.000A
[350185] [SENSOR] CSE7766 @ HwSerial - voltage: 0V
[350186] [SENSOR] CSE7766 @ HwSerial - power: 0W
[350187] [SENSOR] CSE7766 @ HwSerial - energy: 0J
[356189] [SENSOR] CSE7766 @ HwSerial - current: 0.000A
[356192] [SENSOR] CSE7766 @ HwSerial - voltage: 0V
[356193] [SENSOR] CSE7766 @ HwSerial - power: 0W
[356195] [SENSOR] CSE7766 @ HwSerial - energy: 0J
[362198] [SENSOR] CSE7766 @ HwSerial - current: 0.000A

xoseperez, 2018-05-25:
(@ServerCare I edited your previous comment so the debug log is visible)

That log is partial, it should be dumping 24 bytes with each line. Best way to do it is to use telnet (web interface debug window cannot cope with many data at once). The only almost complete line is the first one (22 bytes out of 24) and the only significant byte there for this issue if byte 20, which is 0x61.

That's the ADJ byte in the CSE7766 and CSE7759B datasheets and it is telling us that the power cycle was not complete at that moment (bit 4 equals 0). But that's not bad, it simply means it had no time to complete a power cycle measurement, but is should report it in a future measurement. That bit to 0 is compatible with having voltage data but no power or current data (if power measurement is not completed power is set to 0 and current is not even checked).

It will be really useful to have complete logs (several lines with 24 bytes without load and with a known load).

ServerCare, 2018-05-25:
I initially had it connected to the USB only, but I did not see anything in Putty. Does the device need to be swiched into a special debug mode?

Secondly: I recall seeing somewhere that you should never connect the USB -and- power at the same time as that may kill your USB?  The problem is that the device will only report loads when the device is connected to power...

wildwiz, 2018-05-25:
Instead of using the serial you can enable the telnet connection from the admin page and access the trace telnetting the device, with no physical connection between PC and the device required.

ServerCare, 2018-05-25:
I hope this is more useful... sorry for the amount of data... This is a session with no load connected:

[879976] [SENSOR] CSE7766 @ HwSerial - current: 0.000A
[879978] [SENSOR] CSE7766 @ HwSerial - voltage: 231V
[879979] [SENSOR] CSE7766 @ HwSerial - power: 0W
[879980] [SENSOR] CSE7766 @ HwSerial - energy: 551J
[880028] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 03 01 00 3D 36 03 6B 37 4C 72 50 E8 4E A4 61 00 6F DD
[880131] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 03 01 00 3D 36 03 6B 37 4C 72 50 E9 E8 09 61 00 6F DD
[880234] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 03 01 00 3D 36 03 6B 37 4C 72 50 EB 82 6D 61 00 6F DD
[880338] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 03 01 00 3D 36 03 6B 37 4C 72 50 ED 1A D3 61 00 6F DD
[880441] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 EE B5 37 61 00 6F DA
[880546] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F0 4E 9C 61 00 6F DA
[880651] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[880763] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[880864] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[880965] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[881076] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[881177] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[881289] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[881390] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[881491] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[881602] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 F1 A4 AA 71 00 70 50
[881703] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 6B 37 4C 72 50 10 3E B1 61 00 70 00
[881804] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 04 C2 26 4C 72 50 11 D8 16 61 00 70 47
[881916] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 04 C2 26 4C 72 50 13 71 7B 61 00 70 47
[882021] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 04 C2 26 4C 72 50 15 0A E0 61 00 70 47
[882124] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 04 C2 26 4C 72 50 16 A3 46 61 00 70 47
[882226] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 04 C2 26 4C 72 50 18 3D AA 61 00 70 47
[882329] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 04 C2 26 4C 72 50 19 D7 0F 61 00 70 47
[882433] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 04 C2 26 4C 72 50 1B 70 74 61 00 70 47
[882541] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 04 C2 26 4C 72 50 1D 08 DA 61 00 70 47
[882648] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 1E A3 3E 61 00 70 49
[882755] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 20 3C A3 61 00 70 49
[882851] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 21 D5 09 61 00 70 49
[882957] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 23 6F 6D 61 00 70 49
[883061] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 25 07 D3 61 00 70 49
[883167] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 26 A1 38 61 00 70 49
[883275] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 28 3B 9C 61 00 70 49
[883379] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 29 D5 01 61 00 70 49
[883485] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 2B 6E 66 61 00 70 49
[883591] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 2D 06 CC 61 00 70 49
[883696] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 2E A1 30 61 00 70 49
[883801] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 30 3A 95 61 00 70 49
[883897] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 31 D3 FA 61 00 70 48
[884011] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 04 C2 26 4C 72 50 33 6D 5F 61 00 70 49
[884107] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 06 B6 09 4C 72 50 35 06 C4 61 00 70 22
[884220] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 06 B6 09 4C 72 50 36 A0 29 61 00 70 22
[884326] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 06 B6 09 4C 72 50 38 39 8E 61 00 70 22
[884430] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 06 B6 09 4C 72 50 39 D1 F4 61 00 70 21
[884535] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 06 B6 09 4C 72 50 3B 6C 58 61 00 70 22
[884630] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 06 B6 09 4C 72 50 3D 04 BE 61 00 70 22
[884735] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 06 B6 09 4C 72 50 3E 9F 22 61 00 70 22
[884843] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 03 00 00 3D 36 06 B6 09 4C 72 50 40 38 87 61 00 70 22
[... cut ...]
[891550] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 7A C8 4C 72 50 A6 91 C7 61 00 70 9F
[891656] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 7A C8 4C 72 50 A8 2A 2D 61 00 70 A0
[891762] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 7A C8 4C 72 50 A9 C4 91 61 00 70 9F
[891869] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 7A C8 4C 72 50 AB 5C F7 61 00 70 9F
[891975] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FF 00 3D 36 03 7A C8 4C 72 50 AC F7 5B 61 00 70 9F

ServerCare, 2018-05-25:
And this is with a small load connected; 

048154] [SENSOR] CSE7766 @ HwSerial - current: 0.000A
[048156] [SENSOR] CSE7766 @ HwSerial - voltage: 234V
[048157] [SENSOR] CSE7766 @ HwSerial - power: 0W
[048158] [SENSOR] CSE7766 @ HwSerial - energy: 601J
[048174] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 CB 99 4F 61 00 79 1C
[048276] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 CD 31 B5 61 00 79 1C
[048377] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 CE CB 1A 61 00 79 1C
[048479] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 D0 65 7E 61 00 79 1C
[048592] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 D1 FE E3 61 00 79 1B
[048694] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 D3 98 48 61 00 79 1C
[048797] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 D5 30 AE 61 00 79 1C
[048899] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 D6 CB 12 61 00 79 1C
[049011] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 D8 63 78 61 00 79 1C
[049114] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 D9 FD DC 61 00 79 1B
[049216] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 DB 96 42 61 00 79 1C
[049317] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 DD 2F A7 61 00 79 1C
[049430] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 DE CA 0B 61 00 79 1C
[049531] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 E0 63 70 61 00 79 1C
[049636] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 E1 FC D5 61 00 79 1B
[049740] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 E3 96 3A 61 00 79 1C
[049843] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 E5 2E A0 61 00 79 1C
[049946] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 E6 C8 05 61 00 79 1C
[050058] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 E8 62 69 61 00 79 1C
[050159] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 E9 FA CF 61 00 79 1B
[050261] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 EB 95 33 61 00 79 1C
[050373] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 ED 2D 99 61 00 79 1C
[050476] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 03 42 C6 4C 72 50 EE C6 FE 61 00 79 1B
[050579] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 04 10 BA 4C 72 50 F0 60 63 61 00 79 DF
[050682] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 04 10 BA 4C 72 50 F1 F9 C8 61 00 79 DE
[050784] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 04 10 BA 4C 72 50 F3 93 2D 61 00 79 DF
[050889] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 04 10 BA 4C 72 50 F5 2C 92 61 00 79 DF
[051002] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 04 10 BA 4C 72 50 F6 C6 F6 61 00 79 DE
[051105] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 04 10 BA 4C 72 50 F8 60 5B 61 00 79 DF
[051208] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 04 10 BA 4C 72 50 F9 F8 C1 61 00 79 DE
[051313] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 F8 00 3D 36 04 10 BA 4C 72 50 FB 93 25 61 00 79 DF
[051416] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FD 2C 8A 61 00 79 E4
[051518] [SENSOR] CSE7766: _process: F2 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE C5 EF 61 00 79 E3
[051631] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 79 AB
[051733] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 7A AC
[051837] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 7A AC
[051942] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 7A AC
[052043] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 7A AC
[052147] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 7A AC
[052259] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 7A AC
[052361] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 7A AC
[052462] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 7A AC
[052566] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 FE F6 76 71 00 7A AC
[052668] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 10 BF 30 61 00 7A 31
[052782] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 12 57 96 61 00 7A 31
[052884] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 13 F0 FB 61 00 7A 30
[052985] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 15 8A 60 61 00 7A 31
[053088] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FD 00 3D 36 04 10 BA 4C 72 50 17 24 C4 61 00 7A 31
[... cut ...]
[059698] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FB 00 3D 36 04 56 91 4C 72 50 7B E3 A0 61 00 7A 4B
[059800] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FB 00 3D 36 04 56 91 4C 72 50 7D 7E 04 61 00 7A 4C
[059904] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FB 00 3D 36 04 56 91 4C 72 50 7F 17 69 61 00 7A 4C
[060006] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FB 00 3D 36 04 56 91 4C 72 50 80 B0 CE 61 00 7A 4B
[060117] [SENSOR] CSE7766: _process: 55 5A 02 B7 50 00 02 FB 00 3D 36 04 56 91 4C 72 50 82 4A 33 61 00 7A 4C

xoseperez, 2018-05-26:
How small is that load? Energy is increasing at 5J/6s, that's a less than 1W load.

xoseperez, 2018-05-26:
I have decoded a couple of lines and here's what I got:
```
55
5A 
02 B7 50    => 178000
00 02 FD    => 765          232.68V
00 3D 36    => 15670
04 10 BA    => 266425       0.06W (?)
4C 72 50    => 5010000
FE F6 76    => 16709238     0.30A
71          => 0111 0001
00 79 
AB

55 
5A 
02 B7 50    => 178000 
00 02 FD    => 765          232.68V 
00 3D 36    => 15670 
04 56 91    => 284305       0.05W (?)
4C 72 50    => 5010000 
62 4E 4F    => 6442575      0.78A
61          => 0110 0001 (no power!!)
00 7A 
4E
```
Looks like at some point it stops reporting power values... I think I have to take a second look at the datasheet...

cent0, 2018-05-27:
@jeanpaul79    for R1 sonoff pow

like post #894 
to continue post #887 

SENSOR_POWER_CHECK_STATUS set to 0       THIS FUNCTION NOT TURN OFF the data if relay is OFF,

i have solves this for version 1.12.5 and news, 

 and have same problem for setting new password .... post #895 
have rebuild over 10 new bin and not working :( :+1:

ServerCare, 2018-05-28:
@xoseperez I think you are on to something.... it looks like the load was just too small. The fact that joules were showing an increasing number was throwing me off... The load was a label-printer and apparently it was using so little that it simply didn't show up... (that's probably what you saw happening in the log).

I have now given is a much larger load (a fan) and it is showing me:

```
 Current #0    0.213A   CSE7766 @ HwSerial
Voltage #0    235V CSE7766 @ HwSerial
Active Power #0   42W   CSE7766 @ HwSerial
Energy #0   1258647J   CSE7766 @ HwSerial (since 2018-05-28 16:24:22)
```

(is that joule number correct?)

xoseperez, 2018-05-29:
No problem, it's great to see it's working. Closing :smile:

closed 2018-05-29 by xoseperez