xoseperez/espurna #871: not an issue: CSS optimization

gn0st1c, 2018-05-18:
@xoseperez : 
i thought you may want to see/test this.

i renamed pure css class names and saved about 4k. (61060 index.html.gz)
just replaced "pure-" with "p-"

[www.tar.gz](https://github.com/xoseperez/espurna/files/2017551/www.tar.gz)

xoseperez, 2018-05-19:
Amazing. Same concern as with #870 but it really pays off!

gn0st1c, 2018-05-20:
@xoseperez : that was also my concern but it's worth the trouble.

when you update the pure css files, just run; `grep -rl 'pure-' . | xargs sed -i '' 's/pure-/p-/g'`

xoseperez, 2018-05-21:
@gn0st1c I have added a replace in the gulp file that should take care of replacing "pure-" with "p-" over the single merged file just before the gzipping.
But my numbers are not as spectacular. I just get about 1Kb reduction on the final gzipped file... where is the difference? I have pushed the code anyway.

gn0st1c, 2018-05-21:
@xoseperez : do not push the code.. because gulp will also replace file references but won't rename the files;
```
        <link rel="stylesheet" href="vendor/pure-1.0.0.min.css" />
        <link rel="stylesheet" href="vendor/pure-grids-responsive-1.0.0.min.css" />
```

gn0st1c, 2018-05-21:
your and my file size differences are because, i have more optimization locally. removing unnecessary html end tags, etc etc. i'll PR when i'm done :)

xoseperez, 2018-05-21:
@gn0st1c I have pushed my code which does the replacing over the merged file, so the files in the `html` folder are not changed.

closed 2018-05-22 by xoseperez