xoseperez/espurna #872: MQTT Message Boot state Problem

Mattleyu30, 2018-05-19:
Hello everyone
a few days ago I had a problem of disconnection of energy in an area of ​​the house
where I had sonoff devices with espurna connected with Domoticz.
These sonoffs had an off time schedule, so domoticz continued to work because on another home power line and the espurna sonoff were off. Before the energy returned, the Domoticz schedule turned off the devices but the MQTT message could not be reached. On the return of energy, Sonoff Espurna resumed working in the state as it was before the energy was not there (ON) but did not send any message MQTT boot state to Domoticz, so I had espurna ON but on domoticz I saw it in OFF . (I realized only the day after because they were water extraction pumps and not lights (visible).
Question:
1 ° how is it possible that espurna send a boot state when it is switched boot?
2 ° to avoid any communication errors, perhaps due to the range of the Wi-fi, is it possible that Espurna sends a MQTT Message of the status of the switches every 3-5 minutes, as it does for the sensors?
This allows you to always have control of the current state of the switches.

Version Espurna 1.12.6

Thank

xoseperez, 2018-05-19:
The heartbeat messages do have the relay status by default (that's the HEARTBEAT_REPORT_RELAY setting in general.h). They are sent every 5 minutes by default.

Mattleyu30, 2018-05-21:
Hi, Thanks for the reply
I did a test repeating the shutdown of the Espurna sonoffs and leaving Domoticz active but switching off the switch.
When the reactivation of the energy at the Sonoff expires, the relay is returned to ON but even after 5 minutes it has not sent any MQTT message to the Domoticz of the state.
I tried to see even in the debug but it does not pass any message to the relay.
I need to change some options in the general.h file, or on the web interface to activate Heartbeat.

xoseperez, 2018-05-21:
The heartbeat is active by default and should report the status every 5 minutes (along other values). That, assuming you have MQTT enabled in the web UI (using the same broker as Domoticz). Also, check you don't have "Use JSON payload" enabled since this is not compatible with Domoticz.

Mattleyu30, 2018-05-21:
the broker and mosquito mounted on raspberry pi 3 where there is also domoticz mounted. the broker is only 1, and I also see all the messages of other devices in the debug of espurna. Also "use pay load". I have no idea why it does not update the status.

Mattleyu30, 2018-05-22:
hi, after a factory reset heartbeat began to work. But Domoticz has not interpreted the command I guess because there is no IDX switch. How can I make it talk to domoticz.
Debug log script
[906262] [MQTT] Sending ESPURNA_581D6B / app => EXPURNA (PID 1)
[906264] [MQTT] Sending ESPURNA_581D6B / version => 1.12.6 (PID 1)
[906266] [MQTT] Sending ESPURNA_581D6B / board => ITEAD_SONOFF_TH (PID 1)
[906270] [MQTT] Sending ESPURNA_581D6B / host => ESPURNA_581D6B (PID 1)
[906276] [MQTT] Sending ESPURNA_581D6B / ip => 192.168.200.241 (PID 1)
[906282] [MQTT] Sending ESPURNA_581D6B / mac => 5C: CF: 7F: 58: 1D: 6B (PID 1)
[906288] [MQTT] Sending ESPURNA_581D6B / rssi => -60 (PID 1)
[906293] [MQTT] Sending ESPURNA_581D6B / uptime => 906 (PID 1)
[906299] [MQTT] Sending ESPURNA_581D6B / datetime => 2018-05-22 09:43:53 (PID 1)
[906305] [MQTT] Sending ESPURNA_581D6B / freeheap => 22512 (PID 1)
[906311] [MQTT] Sending ESPURNA_581D6B / relay / 0 => 1 (PID 1)
[906317] [MQTT] Sending ESPURNA_581D6B / vcc => 3142 (PID 1)
[906321] [MQTT] Sending ESPURNA_581D6B / status => 1 (PID 1)
[906326] [MQTT] Sending ESPURNA_581D6B / loadavg => 1 (PID 1)

Mattleyu30, 2018-05-24:
Hi Xose
the change to the relay code works regularly both on boot and when it loses the wi-fi connection.
at this point I close the message.

Thank you for your support

closed 2018-05-24 by Mattleyu30