xoseperez/espurna #873: unable to factory reset Sonoff RF Bridge R2

zerog2k, 2018-05-19:
I have an RFBridge (rev 2?) which was working ok until recently. I tried connecting to webui, but I forgot password. Now I'm unable to reset password by returning to factory defaults, i.e. by holding the button ("pairing button") down for >10 seconds.

Not sure how to reset this without completely reflashing.

zerog2k, 2018-05-19:
Ok, so tearing my hair out for about an hour, reflashing to latest code, etc etc. Was not able to access til I switched over to chrome browser on windows 10 laptop. (I was trying in vain with Edge browser. :(

Found that it seems not only Safari, but also Windows10/Edge browser does not work. 

The auth is somehow not compatible with enough browsers. Not sure how to fix this, but in factory defaults, the unit should come up with the _most_ compatible settings as possible, for general UX reasons.

So not sure if it actually was a factory reset problem or not.

zerog2k, 2018-05-19:
I added a note to troubleshooting section of wiki.
https://github.com/xoseperez/espurna/wiki/Troubleshooting/_compare/f30f53fdb395d9e380eee1da9538d8fa4c7163a2

zerog2k, 2018-05-23:
I wonder if it's possible to either disable WSAuth by default and give user to option enable it during initial password setting (with note about compatible/incompatible browsers), or determine this setting dynamically during auth, e.g. by using some regex of supported/unsupported browsers.

xoseperez, 2018-06-12:
Latest version in dev branch uses the old authentication procedure based on HTTP requests, no WSAuth. Still, I can only test with FF and Chrome for Linux. @zerog2k, do you mind giving it a try?

xoseperez, 2018-06-22:
Released with 1.13.0. Closing.

closed 2018-06-22 by xoseperez